The-OpenROAD-Project/OpenROAD-flow-scripts #478: Cannot run a gcd sample design: Error 2

AyaseErii, 2022-06-05:
**Describe the bug**
I am new to OpenROAD flow. I followed the tutorial to install the openroad flow locally. However, I cannot even run the sample gcd design after I remove the comment sign of the design. An error shows as below: 

    [INFO][FLOW] Using platform directory ./platforms/nangate45
    /bin/sh: 0: Illegal option -o pipefail
    /bin/sh: 0: Illegal option -o pipefail
    /bin/sh: 0: Illegal option -o pipefail
    /bin/sh: 0: Illegal option -o pipefail
    /bin/sh: 0: Illegal option -o pipefail
    /bin/sh: 0: Illegal option -o pipefail
    /bin/sh: 0: Illegal option -o pipefail
    /bin/sh: 0: Illegal option -o pipefail
    /bin/sh: 0: Illegal option -o pipefail
    /bin/sh: 0: Illegal option -o pipefail
    make: *** [Makefile:311: objects/nangate45/gcd/base/lib/NangateOpenCellLibrary_typical.lib] Error 2

**To reproduce the error**
* Run the script below as shown in the tutorial:

      git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
      cd OpenROAD-flow-scripts
      ./build_openroad.sh --local
      source ./setup_env.sh

* Enter the flow dictionary, remove the comment of gcd design as `DESIGN_CONFIG=./designs/nangate45/gcd/config.mk` and then run `make`. You will get the error mentioned above.

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: [Ubuntu 20.04]
 - OpenROAD-flow-scripts latest

**Additional context**
Is there anything I missed? Could anyone point me to a right direction? I will appreciate it!

vvbandeira, 2022-06-05:
Please see https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/330 for a possible solution.

AyaseErii, 2022-06-05:
> Please see #330 for a possible solution.

Thanks for your help! Here is a one-sentence summary of solving this issue: remove the `?` in line 102 of the MakeFile. Just in case someone meets with the same problem.

closed 2022-06-05 by vvbandeira