The-OpenROAD-Project/OpenROAD-flow-scripts #480: Error1 in the step of 3_3_place_gp

AyaseErii, 2022-06-10:
**Describe the bug**
I am trying to use OpenROAD-flow to generate the GDSII of a custom design. The tool stuck on the step `3_3_place_gp` while the step1 yosys synthesis and step2 floorplan look good to me. The error of step3 shows as below:

      [NesterovSolve] Iter: 290 overflow: 0.785725 HPWL: 103478460
      Stack trace:
       0# 0x0000557545D42511 in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
       1# 0x00007F472CA620C0 in /lib/x86_64-linux-gnu/libc.so.6
       2# rsz::BufferedNet::BufferedNet(rsz::BufferedNetType, odb::Point, std::shared_ptr<rsz::BufferedNet>, std::shared_ptr<rsz::BufferedNet>) in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
       3# 0x0000557545EAE53F in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
       4# 0x0000557545EAE464 in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
       5# 0x0000557545EAE464 in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
       6# rsz::Resizer::makeBufferedNetSteiner(sta::Pin const*, sta::Corner const*) in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
       7# rsz::RepairDesign::repairNet(sta::Net*, sta::Pin const*, sta::Vertex*, double, double, bool, bool, bool, int, bool, int&, int&, int&, int&, int&) in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
       8# rsz::RepairDesign::repairDesign(double, double, double, int&, int&, int&, int&, int&) in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
       9# rsz::Resizer::findResizeSlacks() in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
      10# gpl::TimingBase::updateGNetWeights(float) in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
      11# gpl::NesterovPlace::doNesterovPlace(int) in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
      12# 0x0000557545D9B5D8 in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
      13# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
      14# 0x00007F472E79F924 in /lib/x86_64-linux-gnu/libtcl8.6.so
      15# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
      16# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
      17# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
      18# ord::tclAppInit(Tcl_Interp*) in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
      19# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
      20# main in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
      21# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
      22# _start in /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
      Command exited with non-zero status 1
      Elapsed time: 0:03.46[h:]min:sec. CPU time: user 3.40 sys 0.03 (99%). Peak memory: 178940KB.
      make: *** [Makefile:488: results/sky130hd/mydesign/base/3_3_place_gp.odb] Error 1

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS:  Ubuntu 20.04.3 LTS
 - OpenROAD-flow latest

**Additional context**
The build-in example designs are working well. Could anyone tell me how to debug this error? I will appreciate it! Thanks in advance!

vvbandeira, 2022-06-10:
Can you try with the latest `tools/OpenROAD`? There was a similar issue fixed upstream recently.

```
./build_openroad.sh --local --latest
```

Or if you are using docker.
```
./build_openroad.sh --latest
```

AyaseErii, 2022-06-10:
Thanks for your reply, Vitor. I built the tool locally, and I tried the command:
`./build_openroad.sh --local --latest`
It showed another error when it processed 86%. The error message is shown below:

    [ 86%] Built target rcx
    [ 86%] Linking CXX static library gui.a
    cd /home/jun/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gui && /usr/local/bin/cmake -P CMakeFiles/gui.dir/cmake_clean_target.cmake
    cd /home/jun/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gui && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/gui.dir/link.txt --verbose=1
    /usr/bin/ar qc gui.a  CMakeFiles/gui.dir/gui_autogen/mocs_compilation.cpp.o CMakeFiles/gui.dir/CMakeFiles/gui.dir/guiTCL_wrap.cxx.o CMakeFiles/gui.dir/gui-tclInitVar.cc.o CMakeFiles/gui.dir/src/layoutViewer.cpp.o CMakeFiles/gui.dir/src/mainWindow.cpp.o CMakeFiles/gui.dir/src/scriptWidget.cpp.o CMakeFiles/gui.dir/src/tclCmdInputWidget.cpp.o CMakeFiles/gui.dir/src/tclCmdHighlighter.cpp.o CMakeFiles/gui.dir/src/displayControls.cpp.o CMakeFiles/gui.dir/src/gui.cpp.o CMakeFiles/gui.dir/src/search.cpp.o CMakeFiles/gui.dir/src/findDialog.cpp.o CMakeFiles/gui.dir/src/inspector.cpp.o CMakeFiles/gui.dir/src/dbDescriptors.cpp.o CMakeFiles/gui.dir/src/highlightGroupDialog.cpp.o CMakeFiles/gui.dir/src/selectHighlightWindow.cpp.o CMakeFiles/gui.dir/src/staGui.cpp.o CMakeFiles/gui.dir/src/timingWidget.cpp.o CMakeFiles/gui.dir/src/drcWidget.cpp.o CMakeFiles/gui.dir/src/ruler.cpp.o CMakeFiles/gui.dir/src/heatMap.cpp.o CMakeFiles/gui.dir/src/heatMapSetup.cpp.o CMakeFiles/gui.dir/src/heatMapPlacementDensity.cpp.o CMakeFiles/gui.dir/src/browserWidget.cpp.o CMakeFiles/gui.dir/gui_autogen/3YJK5W5UP7/qrc_resource.cpp.o
    /usr/bin/ranlib gui.a
    cd /home/jun/OpenROAD-flow-scripts/tools/OpenROAD/src/gui && /home/jun/OpenROAD-flow-scripts/tools/OpenROAD/etc/find_messages.py > /home/jun/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/messages.txt
    make[2]: Leaving directory '/home/jun/OpenROAD-flow-scripts/tools/OpenROAD/build'
    [ 86%] Built target gui
    make[1]: Leaving directory '/home/jun/OpenROAD-flow-scripts/tools/OpenROAD/build'
    make: *** [Makefile:133: all] Error 2

Did I miss something with gui? I believe I installed the required dependencies for gui and there was no such an error showing up the first time I installed the tool.

vvbandeira, 2022-06-10:
The build runs in parallel so the error is probably earlier in the log / terminal stdout.

Did you clean your workspace? Maybe try the following before building again.
```
git clean -d -x -i
git submodule foreach --recursive git clean -d -x -i
```

On your next build run the following in case it has an error, then you can upload the build.log for analysis.
```
./build_openroad.sh --local --latest 2>&1 | tee build.log
```

AyaseErii, 2022-06-10:
Thank you very much for the help! It did work after following your suggestions! Appreciate it!

vvbandeira, 2022-06-10:
No problem.

(I will mark it as closed after I create a PR to update the `tools/OpenROAD` submodule)

closed 2022-06-12 by vvbandeira