The-OpenROAD-Project/OpenROAD-flow-scripts #483: A routing error of a large design

AyaseErii, 2022-06-13:
**Describe the bug**
Hello, I tried to generate a GDSII for a larger custom design with OpenROAD. Everything went well from 1_yosys synthesis to 4_cts. However, the tool stuck at 5_route with the error (Error137) messages below:

    ==========================================================================
    check_antennas
    --------------------------------------------------------------------------
    [INFO ANT-0001] Found 0 pin violations.
    [INFO ANT-0002] Found 0 net violations in 436064 nets.
    [INFO FLW-0007] clock clk period 1.085200
    [WARNING FLW-0010] more than one clock found. Skipping sdc update.
    [INFO FLW-0011] Path endpoint count 100795
    Elapsed time: 17:24.56[h:]min:sec. CPU time: user 1039.88 sys 0.92 (99%). Peak memory: 2960168KB.
    (/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/jun/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/detail_route.tcl -metrics ./logs/sky130hd/my_design/base/5_2_TritonRoute.json) 2>&1 | tee ./logs/sky130hd/my_design/base/5_2_TritonRoute.log
    OpenROAD v2.0-3976-g43350fce0 
    This program is licensed under the BSD-3 license. See the LICENSE file for details.
    Components of this program may be licensed under more restrictive licenses which must be honored.
    [INFO ORD-0030] Using 16 thread(s).
    [INFO DRT-0149] Reading tech and libs.
    
    Units:                1000
    Number of layers:     13
    Number of macros:     441
    Number of vias:       25
    Number of viarulegen: 25
    
    [INFO DRT-0150] Reading design.
    
    Design:                   my_design_top
    Die area:                 ( 0 0 ) ( 2691190 2691190 )
    Number of track patterns: 12
    Number of DEF vias:       4
    Number of components:     1119600
    Number of terminals:      196
    Number of snets:          2
    Number of nets:           436064
    
    [INFO DRT-0167] List of default vias:
      Layer mcon
        default via: L1M1_PR
      Layer via
        default via: M1M2_PR
      Layer via2
        default via: M2M3_PR
      Layer via3
        default via: M3M4_PR
      Layer via4
        default via: M4M5_PR
    [INFO DRT-0162] Library cell analysis.
    [INFO DRT-0163] Instance analysis.
      Complete 10000 instances.
      Complete 20000 instances.
      Complete 30000 instances.
      Complete 40000 instances.
      Complete 50000 instances.
      Complete 60000 instances.
      Complete 70000 instances.
      Complete 80000 instances.
      Complete 90000 instances.
      Complete 100000 instances.
      Complete 200000 instances.
      Complete 300000 instances.
      Complete 400000 instances.
      Complete 500000 instances.
      Complete 600000 instances.
      Complete 700000 instances.
      Complete 800000 instances.
      Complete 900000 instances.
      Complete 1000000 instances.
      Complete 1100000 instances.
    [INFO DRT-0164] Number of unique instances = 58.
[... 211 lines cut ...]
    [INFO DRT-0036] met4 guide region query size = 8391.
    [INFO DRT-0036] via4 guide region query size = 0.
    [INFO DRT-0036] met5 guide region query size = 1.
    [INFO DRT-0179] Init gr pin query.
    [INFO DRT-0185] Post process initialize RPin region query.
    [INFO DRT-0181] Start track assignment.
    [INFO DRT-0184] Done with 1236907 vertical wires in 8 frboxes and 640667 horizontal wires in 8 frboxes.
    [INFO DRT-0186] Done with 182652 vertical wires in 8 frboxes and 167023 horizontal wires in 8 frboxes.
    [INFO DRT-0182] Complete track assignment.
    [INFO DRT-0267] cpu time = 00:02:44, elapsed time = 00:00:25, memory = 9807.75 (MB), peak = 9942.76 (MB)
    [INFO DRT-0187] Start routing data preparation.
    [INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 9807.75 (MB), peak = 9942.76 (MB)
    [INFO DRT-0194] Start detail routing.
    [INFO DRT-0195] Start 0th optimization iteration.
    Command terminated by signal 9
    Elapsed time: 1:46.13[h:]min:sec. CPU time: user 998.84 sys 10.45 (950%). Peak memory: 12510284KB.
    make: *** [Makefile:575: results/sky130hd/my_design/base/5_route.odb] Error 137

**Environment (please complete the following information):**
Please only report issues for supported OSs.
- OS: Ubuntu 20.04.3 LTS
- OpenROAD-flow latest

**Additional context**
The tool works with a relatively small custom design but cannot work with large design. Is there anything I missing? Could anyone help me with fixing this error issue? I will appreciate it!

vijayank88, 2022-06-13:
@AyaseErii 
Run `make detail_routing_issue` in the `flow` directory and attach issue reproducible test case.

What's system RAM size?

vijayank88, 2022-06-13:
try to increase swap space and try once.

@vvbandeira `terminated by signal 9` is out of memory kill right?

AyaseErii, 2022-06-14:
Thanks for the explanation. I think it is a memory issue. I checked the the RAM and increased the swap space. Here is the info after increasing the swap space and using `sudo free -h` command to see if the space was increased:

                    total        used        free      shared  buff/cache   available
      Mem:           12Gi       1.0Gi        10Gi       0.0Ki       543Mi        11Gi
      Swap:          18Gi       1.0Gi        18Gi

PS: Here I really do not know why the total memory is 12G rather than 16G.

Then I tried the routing again, it can go to `5_2_TritonRoute` stage (the tool cannot go into step 5 at all before increasing the swap space) but still show the error indicating the memory:

    [INFO DRT-0036] met2 guide region query size = 347832.
    [INFO DRT-0036] via2 guide region query size = 0.
    [INFO DRT-0036] met3 guide region query size = 25553.
    [INFO DRT-0036] via3 guide region query size = 0.
    [INFO DRT-0036] met4 guide region query size = 8391.
    [INFO DRT-0036] via4 guide region query size = 0.
    [INFO DRT-0036] met5 guide region query size = 1.
    [INFO DRT-0179] Init gr pin query.
    [INFO DRT-0185] Post process initialize RPin region query.
    [INFO DRT-0181] Start track assignment.
    [INFO DRT-0184] Done with 1236907 vertical wires in 8 frboxes and 640667 horizontal wires in 8 frboxes.
    [INFO DRT-0186] Done with 182652 vertical wires in 8 frboxes and 167023 horizontal wires in 8 frboxes.
    [INFO DRT-0182] Complete track assignment.
    [INFO DRT-0267] cpu time = 00:02:49, elapsed time = 00:00:29, memory = 9797.41 (MB), peak = 10048.95 (MB)
    [INFO DRT-0187] Start routing data preparation.
    [INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 9797.41 (MB), peak = 10048.95 (MB)
    [INFO DRT-0194] Start detail routing.
    [INFO DRT-0195] Start 0th optimization iteration.
        Completing 10% with 0 violations.
        elapsed time = 00:01:13, memory = 11914.71 (MB).
    Command terminated by signal 9
    Elapsed time: 3:46.00[h:]min:sec. CPU time: user 2348.75 sys 36.22 (1055%). Peak memory: 12497756KB.

Do I need to add more memory to my local machine? Any suggestions?

Thanks!

vijayank88, 2022-06-14:
@AyaseErii 
Yes need to increase your RAM.
Is that Docker or local installation?

AyaseErii, 2022-06-14:
I see. That is local installation, not docker.

vijayank88, 2022-06-14:
I don't understan why `make detail_routing_issue` not working to provide the test case.

Try with increased system resource.

vvbandeira, 2022-06-14:
> PS: Here I really do not know why the total memory is 12G rather than 16G.

I have seen similar things if the system does not have a dedicated GPU, that memory can be allocated to the on-board graphics.

@AyaseErii can you share a reproducible testcase so we can try to run on our servers? Then we can be sure if it is a resource issue and how much you will need.

if you cannot share, I would suggest adding as much swap as you can and running again. Note that using swap will increase runtime.

AyaseErii, 2022-06-15:
I am trying to use a server with a larger memory to re-do the same design. The tool is running now and it is in `cts` stage. I will let you guys know once the flow was done.

vijayank88, 2022-06-16:
@AyaseErii 
Is memory issue got fixed?

vvbandeira, 2022-06-16:
That's great @AyaseErii, please let us know if the design finishes the flow without error with the increased memory so we can close the issue.

AyaseErii, 2022-06-16:
Thank you very much for the help. The tool finished the design without error.

closed 2022-06-16 by vvbandeira