The-OpenROAD-Project/OpenROAD-flow-scripts #491: Directory not found when trying to run the flow

GabbedT, 2022-06-17:
Hi, i'm trying to synthetize my first custom design. I created two directories: the source files in /src/UART/ and the configuration files in /design/sky130hd/UART.
I also modified the Makefile by adding the following line uncommented: DESIGN_CONFIG=./designs/sky130hd/UART/config.mk.

 Then i tried to run: `make DESIGN_CONFIG=./designs/sky130hd/UART/config.mk` and the following output appeared:
 ```
Makefile:128: designs/sky130hd/UART/config.mk: No such file or directory
[INFO][FLOW] Using platform directory ./platforms/
Makefile:143: platforms//config.mk: No such file or directory
make: *** No rule to make target `platforms//config.mk'.  Stop.
``` 

To me it seems like that the directories aren't updated. 

This is the sequence of commands i use to run OpenROAD:

```
 cd OpenROAD-flow-scripts

xhost +local:docker

docker \
   run \
   --tty \
   --interactive \
   --user $(id -u ${USER}):$(id -g ${USER}) \
   --net=host \
   --ipc=host \
   --env DISPLAY=${DISPLAY} \
   --volume $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro \
   --volume $(pwd)/flow/scripts:/OpenROAD-flow-scripts/flow/scripts:ro \
   --volume ${HOME}:/home/${USER} \
   --volume /tmp/.X11-unix:/tmp/.X11-unix \
   --volume ${HOME}/.Xauthority:/.Xauthority \
   --security-opt seccomp=unconfined \
   --privileged \
   openroad/flow-scripts \
   bash
   
source setup_env.sh

cd flow
```

I also tried: `make clean_all`

tspyrou, 2022-06-17:
I think it best that you use the OpenLane setup for your work on sky130. It is the main design script repo for this PDK.
In the meantime in openroad-flow-scripts you can try running the prepackaged aes design. If that works compare your makefile to the makefile in that design directory.

maliberty, 2022-06-17:
I guess you config is missing

export PLATFORM    = sky130hd

but you should post it if that doesn't fix the issue.

GabbedT, 2022-06-17:
@maliberty This is my config.mk file:
```
export DESIGN_NICKNAME = UART
export DESIGN_NAME = uart
export PLATFORM    = sky130hd

export VERILOG_FILES = export VERILOG_FILES = $(sort $(wildcard ./designs/src/$(DESIGN_NICKNAME)/*.v))				   

export SDC_FILE = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/constraint.sdc

# Adders degrade ibex setup repair
export ADDER_MAP_FILE :=

export CORE_UTILIZATION = 70 
export CORE_ASPECT_RATIO = 1
export CORE_MARGIN = 2
export PLACE_DENSITY = 0.65

# Synthesis strategies
export ABC_AREA = 1

export FASTROUTE_TCL = $(PLATFORM_DIR)/fastroute_base.tcl

# IR drop estimation supply net name to be analyzed and supply voltage variable
# For multiple nets: PWR_NETS_VOLTAGES  = "VDD1 1.8 VDD2 1.2"
export PWR_NETS_VOLTAGES  = "VDD 1.8"
export GND_NETS_VOLTAGES  = "VSS 0.0"
```

GabbedT, 2022-06-17:
@tspyrou The other designs work perfectly. Also what are the advantages of OpenLane over OpenROAD?

vvbandeira, 2022-06-17:
My guess is that you generated the image before adding your design files, and the `--volume` parameters do not account for this use case. Can you try the following?

```
docker \
   run \
   --tty \
   --interactive \
   --user $(id -u ${USER}):$(id -g ${USER}) \
   --net=host \
   --ipc=host \
   --env DISPLAY=${DISPLAY} \
   --volume $(pwd)/flow:/OpenROAD-flow-scripts/flow \
   --volume ${HOME}:/home/${USER} \
   --volume /tmp/.X11-unix:/tmp/.X11-unix \
   --volume ${HOME}/.Xauthority:/.Xauthority \
   --security-opt seccomp=unconfined \
   --privileged \
   openroad/flow-scripts \
   bash
```

GabbedT, 2022-06-17:
@vvbandeira Now when i run the flow:
```
bash-4.2$ make DESIGN_CONFIG=./designs/sky130hd/UART/config.mk
[INFO][FLOW] Using platform directory ./platforms/sky130hd
make: *** No rule to make target `export', needed by `results/sky130hd/UART/base/1_1_yosys.v'.  Stop.
```

I noticed that i do not have any `result` directory: `Makefile  designs  objects  platforms  scripts	test  tutorials  util`, this is the  flow directory

vvbandeira, 2022-06-17:
@GabbedT can you fork ORFS, add/commit your changes and provide a link? Is very hard to debug make issues without access to all the relevant files.

GabbedT, 2022-06-17:
@vvbandeira https://github.com/GabbedT/OpenROAD-flow-scripts

vvbandeira, 2022-06-18:
> ```
> export VERILOG_FILES = export VERILOG_FILES = $(sort $(wildcard ./designs/src/$(DESIGN_NICKNAME)/*.v))				   
> ```
> should probably be
> ```
> export VERILOG_FILES = $(sort $(wildcard ./designs/src/$(DESIGN_NICKNAME)/*.v))				   
> ```

I still see this problem [here](https://github.com/GabbedT/OpenROAD-flow-scripts/blob/9db7fa5e1bdea00c7540795cb63e0e1382b1ea1e/flow/designs/sky130hd/UART/config.mk#L5).

Also, there are a couple of commits in your fork deleting essential folders. Example: `tools/` [here](6c99926c684a2ceeefdc87d07bc15d11c4e1f36f).

Steps I ran:
- Clone your repo
- Restore the commits that have the message "Delete X"
- `./build_openroad.sh`
- Run the docker command from https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/491#issuecomment-1158994064
- `cd flow`
- `make DESIGN_CONFIG=./designs/sky130hd/UART/config.mk`

The output I got looks like are issues with Yosys / your design:
```
bash-4.2$ make DESIGN_CONFIG=./designs/sky130hd/UART/config.mk
[INFO][FLOW] Using platform directory ./platforms/sky130hd
./util/markDontUse.py -p "sky130_fd_sc_hd__probe_p_8 sky130_fd_sc_hd__probec_p_8 sky130_fd_sc_hd__lpflow_bleeder_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_16 sky130_fd_sc_hd__lpflow_clkbufkapwr_2 sky130_fd_sc_hd__lpflow_clkbufkapwr_4 sky130_fd_sc_hd__lpflow_clkbufkapwr_8 sky130_fd_sc_hd__lpflow_clkinvkapwr_1 sky130_fd_sc_hd__lpflow_clkinvkapwr_16 sky130_fd_sc_hd__lpflow_clkinvkapwr_2 sky130_fd_sc_hd__lpflow_clkinvkapwr_4 sky130_fd_sc_hd__lpflow_clkinvkapwr_8 sky130_fd_sc_hd__lpflow_decapkapwr_12 sky130_fd_sc_hd__lpflow_decapkapwr_3 sky130_fd_sc_hd__lpflow_decapkapwr_4 sky130_fd_sc_hd__lpflow_decapkapwr_6 sky130_fd_sc_hd__lpflow_decapkapwr_8 sky130_fd_sc_hd__lpflow_inputiso0n_1 sky130_fd_sc_hd__lpflow_inputiso0p_1 sky130_fd_sc_hd__lpflow_inputiso1n_1 sky130_fd_sc_hd__lpflow_inputiso1p_1 sky130_fd_sc_hd__lpflow_inputisolatch_1 sky130_fd_sc_hd__lpflow_isobufsrc_1 sky130_fd_sc_hd__lpflow_isobufsrc_16 sky130_fd_sc_hd__lpflow_isobufsrc_2 sky130_fd_sc_hd__lpflow_isobufsrc_4 sky130_fd_sc_hd__lpflow_isobufsrc_8 sky130_fd_sc_hd__lpflow_isobufsrckapwr_16 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_2 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_4 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_4 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_1 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_2 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_4" -i platforms/sky130hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib -o objects/sky130hd/UART/base/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
Opening file for replace: platforms/sky130hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
Marked 36 cells as dont_use
Commented 0 lines containing "original_pin"
Replaced malformed functions 0
Writing replaced file: objects/sky130hd/UART/base/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
mkdir -p ./results/sky130hd/UART/base ./logs/sky130hd/UART/base ./reports/sky130hd/UART/base
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /OpenROAD-flow-scripts/tools/install/yosys/bin/yosys -v 3 -c ./scripts/synth.tcl) 2>&1 | tee ./logs/sky130hd/UART/base/1_1_yosys.log
1. Executing Verilog-2005 frontend: ./designs/src/UART/baud_rate_generator.v
2. Executing Verilog-2005 frontend: ./designs/src/UART/configuration_registers.v
Warning: Yosys has only limited support for tri-state logic at the moment. (./designs/src/UART/configuration_registers.v:490)
3. Executing Verilog-2005 frontend: ./designs/src/UART/control_unit.v
4. Executing Verilog-2005 frontend: ./designs/src/UART/edge_detector.v
5. Executing Verilog-2005 frontend: ./designs/src/UART/interrupt_arbiter.v
Warning: Yosys has only limited support for tri-state logic at the moment. (./designs/src/UART/interrupt_arbiter.v:157)
Warning: Yosys has only limited support for tri-state logic at the moment. (./designs/src/UART/interrupt_arbiter.v:158)
6. Executing Verilog-2005 frontend: ./designs/src/UART/receiver.v
./designs/src/UART/receiver.v:1: ERROR: Re-definition of module `$abstract\sync_FIFO_buffer'!
Elapsed time: 0:00.05[h:]min:sec. CPU time: user 0.03 sys 0.01 (96%). Peak memory: 6848KB.
make: *** [results/sky130hd/UART/base/1_1_yosys.v] Error 1
```

GabbedT, 2022-06-18:
@vvbandeira Ok, now my output is the same. When i converted the .sv files into .v the FIFO module was translated into both `receiver.v` and `transmitter.v`. Then I created another file called `sync_FIFO_buffer.v` and deleted the previous redefinitions. Now this is the error:
```
7. Executing Verilog-2005 frontend: ./designs/src/UART/sync_FIFO_buffer.v
./designs/src/UART/sync_FIFO_buffer.v:1: ERROR: Re-definition of module `$abstract\sync_FIFO_buffer'!
Elapsed time: 0:00.01[h:]min:sec. CPU time: user 0.01 sys 0.00 (100%). Peak memory: 10372KB.
make: *** [results/sky130hd/UART/base/1_1_yosys.v] Error 1
```
This is the updated folder: https://github.com/GabbedT/OpenROAD-flow-scripts/tree/master/flow/designs/src/UART

I searched the error but the only thing i found that is related to my problem is this: https://github.com/YosysHQ/yosys/issues/732

vijayank88, 2022-06-18:
> @tspyrou The other designs work perfectly. Also what are the advantages of OpenLane over OpenROAD?

@GabbedT 
OpenROAD will do floorplan to routing. The other signoff check done by OpenLane flow.
Install OpenLane from here https://github.com/The-OpenROAD-Project/OpenLane
Also with your UART/src file I'm able to complete the GDS generation will OpenLane. Try and let us know

GabbedT, 2022-06-18:
@vijayank88 I'll try that out. Also, is it possible to visualize the synthetized design with the OpenROAD gui, i found that really helpful for synthesis exploration for the various options like visualizing heat maps, clock trees etc.

vijayank88, 2022-06-18:
@GabbedT 
yes you can get all features. OpenLane is an integrated platform of OpenROAD and other open source tools.

GabbedT, 2022-06-18:
@vijayank88 Thanks, then i'll definitely move to OpenLane flow, can I close the issue?

closed 2022-06-18 by vvbandeira