The-OpenROAD-Project/OpenROAD-flow-scripts #492: Feature of hierarchical floor planning

AyaseErii, 2022-06-17:
**Is your feature request related to a problem? Please describe.**
Hello, I am wondering if the OpenROAD has a feature of hierarchical floor planning. For example, I create a single element layout first and then place the elements in an array, and finally do a routing. Because it takes a long time to perform a global P&R to generate a large GDSII layout, I guess I can use a hierarchical floor planning method with the OpenROAD to reduce the runtime and the area, but I do not know how to do it. 

Is there a feature for that? Do I need to change the floor plan .tcl file? If yes, how do I change it? Are there any examples of hierarchical floor planning in OpenROAD? I would love to contribute to it if needed.

Thanks!

maliberty, 2022-06-18:
We have limited support for this.  We have a basic LEF generator and are bringing out a simple timing model generator.   Hierarchical implementation with a flat signoff is what we recommend for now.

maliberty, 2022-06-18:
OpenLane has some support for this as well using Magic for LEF generation.

vijayank88, 2022-06-18:
@AyaseErii 
Are you talking about macro hardening? 

If yes it may helps https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/hardening_macros.md

AyaseErii, 2022-06-18:
@maliberty Thanks for the suggestion! @vijayank88 Yes, it is exactly what I want! I will take a look in depth. Thank you very much!

closed 2022-06-20 by vijayank88