The-OpenROAD-Project/OpenROAD-flow-scripts #521: Error in building Openroad-flow-scripts locally

Bilel-Manai, 2022-07-18:
Hi all I am trying to build Openroad-flow-scripts locally in a centos7 I get this error meanwhile I have installed it on both ubuntu 20.04 and 22.04 and never seen this error
[image]

vvbandeira, 2022-07-18:
What was the command you used to compile? Also, could you try with a clean workspace? This looks like the tools/install folder got deleted after you called the build script.

Bilel-Manai, 2022-07-18:
I used ./build_openroad.sh --local and I  tried a clean build if that's what you meant by a clean workspace

vvbandeira, 2022-07-18:
I am not sure what may be the issue here since we use Centos 7 for our CI and there all is working as expected. Maybe try to clone the repo again or check you bash setup.

Bilel-Manai, 2022-07-18:
Okay thank you if I find something out I will let you know