The-OpenROAD-Project/OpenROAD-flow-scripts #522: PDN failes on an sram design

Bilel-Manai, 2022-07-18:
Hi all , I am trying to use Openroad latest version on an Sram design there is the design rtl and the config file. 
[sram_v.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9134466/sram_v.txt)
[design_sdc.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9134467/design_sdc.txt)
[harness_mk.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9134468/harness_mk.txt)
 
[image]

Bilel-Manai, 2022-07-18:
Hi 
Thank you for the answer , I can't seem to find the script I am supposed to run 
[image]

Bilel-Manai, 2022-07-18:
[pdn_gcd_nangate45_base_2022-07-18_20-01.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9134621/pdn_gcd_nangate45_base_2022-07-18_20-01.tar.gz)
It's okay thank you for the help !!

maliberty, 2022-07-18:
I ran your test case using the same commit and I get no error:
[image]

Do running the packaged test fail for you?

Bilel-Manai, 2022-07-19:
@vvbandeira yes you are right i did run it again for sram and there is the result and sorry for the confusion.

[pdn_sram_nangate45_base_2022-07-19_11-00.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9139917/pdn_sram_nangate45_base_2022-07-19_11-00.tar.gz)

maliberty, 2022-07-19:
The block is only 11.99 microns wide:
[image]

in the config it specifies
```
add_pdn_stripe -grid {grid} -layer {metal4} -width {0.48} -pitch {56.0} -offset {2}
```
The default spacing between stripes is half the pitch and so your block is too small to insert two stripes.  You could modify your spacing to make it work.  For example this works:
```
add_pdn_stripe -grid {grid} -layer {metal4} -width {0.48} -pitch {56.0} -offset {2} -spacing 1.5
add_pdn_stripe -grid {grid} -layer {metal7} -width {1.40} -pitch {40.0} -offset {2} -spacing 1.5
```
which produces
[image]

maliberty, 2022-07-19:
The design you send earlier had nothing in it but a few tapcells.  I guess the issue is that there is nothing to time in this design.  Where is your logic?

Bilel-Manai, 2022-07-19:
there is the rtl
[msbGeneric.tar.GZ](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9143153/msbGeneric.tar.GZ)

maliberty, 2022-07-19:
Can't open it
```
tar ztf ~/Downloads/msbGeneric.tar.GZ

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
```

Bilel-Manai, 2022-07-19:
can you try and open this one Please 
[msbGeneric.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9143407/msbGeneric.tar.gz)

maliberty, 2022-07-19:
you have no clock in your sdc

Bilel-Manai, 2022-07-19:
silly me I didn't uncomment it but even when I did still same problem

maliberty, 2022-07-19:
Glad to help.  Can this be closed?

closed 2022-07-19 by Bilel-Manai