The-OpenROAD-Project/OpenROAD-flow-scripts #540: global_place_skip_io should reject out of range PLACE_DENSITY

proppy, 2022-07-28:
**Describe the bug**
[RePlAce]
global_place_skip_io currently doesn't reject out of range `PLACE_DENSITY` parameters causing `NesterovSolve` to diverge after 4 hours of iterations and fails with the following error (e.g passing `PLACE_DENSITY` of `80` instead of  `0.8`):
```
[NesterovSolve] Iter: 1380 overflow: 444.77 HPWL: 19022
[ERROR GPL-0305] RePlAce diverged at newStepLength.
double free or corruption (!prev)
Error: global_place_skip_io.tcl, 30 GPL-0305
Stack trace:
```

**Expected behavior**
global_place_skip_io (or RePlAce) validate its input parameters and send errors out before attempting an expensive (and impossible) placement.

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: `Linux f8b4497e4135 4.19.0-21-cloud-amd64 #1 SMP Debian 4.19.249-2 (2022-06-30) x86_64 x86_64 x86_64 GNU/Linux`
 - OpenROAD-flow commit: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/commit/c170615acd5198fd31a03cb975fa9d6781e628e5
 - OpenROAD commit: https://github.com/The-OpenROAD-Project/OpenROAD/commit/e036ecfaca4bc0efe88a54085efcf0f562c48a6b

**File Uploads**
[3_1_place_gp_skip_io.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9213199/3_1_place_gp_skip_io.log)

maliberty, 2022-07-28:
Please package a complete reproducer and not just a log file if possible.  What is the PLACE_DENSITY and what defines it as out of range?

proppy, 2022-07-28:
> What is the PLACE_DENSITY

I used `PLACE_DENSITY = 80` instead of `PLACE_DENSITY = 0.8`

> and what defines it as out of range?

Looking at the existing designs it seems that it is expected to be between `0` and `1`:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/search?q=PLACE_DENSITY+filename%3Aconfig.mk

proppy, 2022-07-28:
> Please package a complete reproducer

Is there an ORFS command for this?

maliberty, 2022-07-28:
make <step>_issue

in this case make global_place_issue

closed 2022-07-29 by maliberty