The-OpenROAD-Project/OpenROAD-flow-scripts #541: createGallery should work with odb file

proppy, 2022-07-29:
**Describe the bug**

`createGallery` seems to expect `.def` file to be present in the `results` directory for all steps, see:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/2c3ed45f6381484ee8fb0501f7c273c91db531eb/flow/util/gallery.json

While the current version of OpenROAD flow only produce `.def` file for a final view:
```
root@f8b4497e4135:/home/jupyter# find runs/20220729005315/results/ | grep def
runs/20220729005315/results/asap7/crc32/base/6_final.def
```

Triggering the following error:
```
make: Entering directory '/OpenROAD-flow-scripts/flow'
[INFO][FLOW] Using platform directory /OpenROAD-flow-scripts/flow/platforms/asap7
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' klayout -z -nc -rx -rd gallery_json=util/gallery.json \
        -rd results_path=/home/jupyter/runs/20220729005315/results/asap7/crc32/base \
        -rd tech_file=/home/jupyter/runs/20220729005315/objects/asap7/crc32/base/klayout.lyt \
        -rm /OpenROAD-flow-scripts/flow/util/createGallery.py) 2>&1 | tee /home/jupyter/runs/20220729005315/logs/asap7/crc32/base/6_1_merge.log
Warning: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Warning: QOpenGLWidget is not supported on this platform.
ERROR: /OpenROAD-flow-scripts/flow/util/createGallery.py:18: Unable to open file: /home/jupyter/runs/20220729005315/results/asap7/crc32/base/2_floorplan.def (errno=2) in MainWindow.load_layout
  /OpenROAD-flow-scripts/flow/util/createGallery.py:18 (class RuntimeError)
Warning: This plugin does not support propagateSizeHints()
```

[Flow]

**Expected behavior**

`png` files are created for each of the flow steps.

**Environment**
 - OS: `Linux f8b4497e4135 4.19.0-21-cloud-amd64 #1 SMP Debian 4.19.249-2 (2022-06-30) x86_64 x86_64 x86_64 GNU/Linux`
 - OpenROAD-flow commit: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/commit/c170615acd5198fd31a03cb975fa9d6781e628e5
 - OpenROAD commit: https://github.com/The-OpenROAD-Project/OpenROAD/commit/e036ecfaca4bc0efe88a54085efcf0f562c48a6b

maliberty, 2022-08-04:
@vvbandeira is createGallery.py used by CI?  @ravi-varadarajan ? I thought this was used for something still.

rovinski, 2022-08-04:
This script was made before the GUI existed (or at least before image saving). The GUI came along and is basically better in every way (don't need a [dedicated script](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/118e0d5c00a8af5e28ddf4f87e356eb1ae46396a/flow/scripts/deleteNonClkNets.tcl) just to delete non-clock nets to get a view of the clock tree)

The only benefit to KLayout over OpenROAD would be to capture the underlying GDS layers which can make some pictures prettier.

vvbandeira, 2022-08-04:
> @vvbandeira is createGallery.py used by CI? @ravi-varadarajan ? I thought this was used for something still.

I don't think `createGallery.py` is used. We have `galleryReport.py`, but that is for the private-tools repo.

maliberty, 2022-08-04:
That must have been what I was thinking of.  If it is truly obsolete we should just remove it.

rovinski, 2022-08-04:
It might be nice to keep the script just for a single GDS screenshot? The other views can be removed.

ravi-varadarajan, 2022-08-04:
Yes we are only using the private_tool_scripts for gallery reports. not
createGallery.py

-ravi

maliberty, 2022-08-04:
If we are only going to keep it for gds snapshots then this issue is moot.