The-OpenROAD-Project/OpenROAD-flow-scripts #550: Merging GDS/OAS file error

lulu9312, 2022-08-05:
Describe the bug
Getting an error at GDS stage during merging GDS/OAS files. The additional files are present in the specified directory. Still it considers only one gds file.

Expected behavior
Should accept all the gds files present.

Screenshots
[image]

rovinski, 2022-08-05:
Please package a test case, it will help to resolve your issue much quicker.

However, all of the cells that are missing are not standard cells. Have you actually set the GDS files in `ADDITIONAL_GDS`?

lulu9312, 2022-08-05:
@rovinski  Yes I have set the GDS files in ADDITIONAL_GDS_FILES. 
Screenshot 
[image]
Here the export GDS command should take all .gds file gds but it considers only one. Here the standard cells are missing as well.

What do you mean pacakage a test case? Should I attach my config file here?

vijayank88, 2022-08-06:
@lulu9312 
ORFS flow don't have built in `sky130_fd_sc_hvl` platforms. Check here: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/flow/platforms.
That's why its shows missing gds. Point  right path to `GDS_FILES`.

@vvbandeira why `hvl` library support missing from ORFS flow for sky130? OpenLane flow has support for `hvl` library.

lulu9312, 2022-08-06:
@vijayank88 Yes I  have already added sky130hvl to the platforms. I have also pointed the right path.

[image]

vijayank88, 2022-08-06:
@lulu9312 
Can you prefix with `ADDITIONAL_LEFS` `ADDITIONAL_GDS` `ADDITIONAL_LIBS`?

lulu9312, 2022-08-07:
@vijayank88 I tried the above prefix  but no change.

lulu9312, 2022-08-08:
@vijayank88 Got the error resolved by updating Klayout.

lulu9312, 2022-08-08:
@vijayank88 Got the error resolved by updating Klayout

closed 2022-08-08 by lulu9312

msaligane, 2022-08-23:
which klayout version have you used? @lulu9312

lulu9312, 2022-08-23:
@msaligane Right now I am on version 0.27.10