The-OpenROAD-Project/OpenROAD-flow-scripts #553: About the final CDL file problem

Elon-J, 2022-08-09:
**Describe the bug**
When I use the OpenRoad script to generate a GDS block from my Verilog design, I find that in the 6_final.cdl file, there are some pins unconnected so that I can't pass the LVS, but I have checked the GDS file and do the post simulation for the GDS block generated by this flow, Its pins connect well and the post-simulation waveform is right. So I suspend that there are some problems in the generation of the CDL file.

**Expected behavior**
All the pins connect rightly in the 6_final.cdl file

**Screenshots**
the '6_final.cdl' file:
[image]
the GDS file corresponds to the cell:
[image]

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: CentOS 7
 - OpenROAD-flow commit: 7719c25a5d991993926677bf7efc4c773b055a7b

**File Uploads**
the cdl_issues:
[cdl_NON_CLK_GEN_sky130hd_base_2022-08-09_22-59.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9295178/cdl_NON_CLK_GEN_sky130hd_base_2022-08-09_22-59.tar.gz)

platform: sky130hd

vijayank88, 2022-08-11:
@Elon-J 
Seems the commits what you're used is old.
Can you try with latest commit and confirm if same issue repeats?

Elon-J, 2022-08-11:
@vijayank88  OK, let me transfer into the newest flow version and try it again. Thanks!

maliberty, 2023-01-11:
no activity - reopen if still relevant with data

closed 2023-01-11 by maliberty