The-OpenROAD-Project/OpenROAD-flow-scripts #555: Autotuner treating successful runs as failed (minimum 9999999999900.0)

ChrisLuck, 2022-08-10:
The root cause of this is that `power__total` is missing in the metrics.json file because commit 3700520d26a556c16c95131ac6447a31d59f6c8f removed the code for that (and a lot more, probably by accident?)

How to reproduce:
`python3 distributed.py --design gcd --platform sky130hd --config designs/sky130hd/gcd/autotuner.json`

ravi-varadarajan, 2022-08-10:
Will look into it now

tspyrou, 2022-09-19:
@vvbandeira when you get back can you look at reverting this change. It break the autotuner.

maliberty, 2022-09-19:
@tspyrou that change is from July.  Surely we have run the autotuner since then...

ChrisLuck, 2022-09-21:
Hi @maliberty !
Maybe I am doing something wrong but I just checked and the problem still exists. The AutoTuner is running without throwing noticeable errors but the results are not useful.

The metric for power is missing so the AutoTuner cannot calculate a correct score. As a result, some "fake" score is returned in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/util/distributed.py#L189
This "fake" score is not a constant but calculated with the step-number so it gives the false impression of a correctly working AutoTuner.

maliberty, 2022-09-21:
I'm restoring the lost power metrics in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2290 and https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/590

tspyrou, 2022-09-22:
@ChrisLuck this is in progress.

maliberty, 2022-09-26:
Both PR have been merged so I think this will work now.  @ahmadelrouby would you test that this fixes the autotuner?

ChrisLuck, 2022-10-10:
I gave it a try and it seems to work without modification now. Thank you very much!

closed 2022-10-10 by vvbandeira