The-OpenROAD-Project/OpenROAD-flow-scripts #572: Add check_setup opensta command to synthesis and final report scripts

tspyrou, 2022-09-01:


vijayank88, 2023-01-16:
@tspyrou 
`check_setup` command added part of `report_metrics`. For `final report` its already available here:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/510b8a1771ecc6c2a7f65508d1753ba7794c4a01/flow/scripts/final_report.tcl#L61
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/510b8a1771ecc6c2a7f65508d1753ba7794c4a01/flow/scripts/report_metrics.tcl#L5

For synthesis stage am not able to add, but possible to add beginning of `floorplan` stage. Shall I add here: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/510b8a1771ecc6c2a7f65508d1753ba7794c4a01/flow/scripts/floorplan.tcl#L3

vijayank88, 2023-01-24:
@maliberty 
Any comment on this?

maliberty, 2023-01-24:
I think floorplan is fine.  It gives us a sense of validity on the original inputs.

vijayank88, 2023-01-24:
Ok thanks for the update. I'll create PR for the same with `check_setup` in floorplan stage.

PR #786 (2023-01-24): added check_setup at floorplan stage

Fixes #572