The-OpenROAD-Project/OpenROAD-flow-scripts #576: [ERROR GPL-0305] RePlAce diverged at newStepLength

vijayank88, 2022-09-08:
**Describe the bug**
A clear and concise description of what the bug is. Please include any tags that apply to this issue:
* [RePlace] for runtime issues 
I'm trying to implement small designs like `digital_pll` and `sar_adc` with less than 100x100 size and failed to complete global placement stage with `absolute` or `relative` size as well.

**Expected behavior**
Complete all stages of the flow.

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: CentOS 7
 - OpenROAD-flow commit : OpenROAD v2.0-4734-ge6d802c6f 

**File Uploads**
[global_place_skip_io_sar_adc_sky130hd_base_2022-09-08_10-33.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9526237/global_place_skip_io_sar_adc_sky130hd_base_2022-09-08_10-33.tar.gz)

**Additional context**
```
[NesterovSolve] Iter: 2520 overflow: 0.19648 HPWL: 3862913
[NesterovSolve] Iter: 2530 overflow: 0.195937 HPWL: 3864635
[NesterovSolve] Iter: 2540 overflow: 0.193654 HPWL: 3882001
[NesterovSolve] Iter: 2550 overflow: 0.195366 HPWL: 3892591
[NesterovSolve] Iter: 2560 overflow: 0.197846 HPWL: 3879648
[NesterovSolve] Iter: 2570 overflow: 0.204245 HPWL: 3899688
[ERROR GPL-0305] RePlAce diverged at newStepLength.
Error: global_place_skip_io.tcl, 30 GPL-0305
Elapsed time: 0:00.80[h:]min:sec. CPU time: user 0.72 sys 0.07 (100%). Peak memory: 82164KB.
make: *** [results/sky130hd/sar_adc/base/3_1_place_gp_skip_io.odb] Error 1
```

joennlae, 2022-09-08:
@vijayank88 I assume it is a small design. Try to add:

```
export HAS_IO_CONSTRAINTS = 1
```
to `config.mk`

vijayank88, 2022-09-09:
@joennlae 
Thanks. Its resolved the issue. 
Where that variable documented?

vijayank88, 2022-09-09:
Also IO pins placed and bottom pins left open. Is that expected?
[image]

joennlae, 2022-09-09:
> @joennlae Thanks. Its resolved the issue. Where that variable documented?

Nowhere. I had to same issue and found out that this is a possible fix for small designs as the solver only works for a certain design size and this option skips the initial placement.

joennlae, 2022-09-09:
> Also IO pins placed and bottom pins left open. Is that expected? [image]

This should be normal as it places IO per default. If I am not mistaken.

maliberty, 2022-09-09:
This is more of a work around than a proper solution.  If the placer diverges we should investigate why.

maliberty, 2022-09-09:
@ramezmoussa the utilization reported here is
```
[INFO GPL-0019] Util(%): 21.11
[INFO GPL-0023] TargetDensity: 0.22
```

Something seems different between how we are computing over
so this should converge easily.  However the overflow stalls
```
[NesterovSolve] Iter: 410 overflow: 0.194881 HPWL: 1468460
...
[NesterovSolve] Iter: 2560 overflow: 0.197846 HPWL: 3879648
```
Something seems off between overflow and utilization/density.  Would you look into how they are both computed?

ramezmoussa, 2022-09-10:
@maliberty  The overall overflow is computed as: $$\sum_i overflow_i \over  \sum standard\textunderscore cell\textunderscore area \_ j + \sum (macro\textunderscore  area \_ j \times targetDensity) $$

where $overflow_i$ is the overflow in bin $i$.  $$overflow_i = \sum \text{(overlap of placable instances with bin i + overlap of fixed instances with bin i)}  - bin\textunderscore area \times targetDensity $$

Regarding utilization, it is computed as $$100 \times (\sum \text{areas of placable instances} \over core\textunderscore area - \sum \text{areas of fixed instances}) $$

Is there anything unusual here?

maliberty, 2022-09-14:
I'm curious why there are there 40 buffers connected in parallel.  Which we will fix the placement issue I am wondering if something is wrong with this design.

vijayank88, 2022-09-22:
@maliberty 
here is source RTL [files](https://github.com/efabless/caravel/blob/main/verilog/rtl/digital_pll.v).

closed 2023-07-17 by vijayank88