The-OpenROAD-Project/OpenROAD-flow-scripts #585: error when running a design

sharkoo7, 2022-09-16:
I build orfs in docker. inside docker:I follow the instructions below:
`source ./setup_env.sh`
`yosys -help`
`openroad -help`
`cd flow`
`make`

But I came across this error:
4.24. Executing ABC pass (technology mapping using ABC).
4.24.1. Extracting gate netlist of module `\gcd' to `<abc-temp-dir>/input.blif'..
ERROR: ABC: execution of command "/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc -s -f /tmp/yosys-abc-jfUrV5/abc.script 2>&1" failed: return code 127.
Elapsed time: 0:00.95[h:]min:sec. CPU time: user 0.74 sys 0.14 (93%). Peak memory: 32440KB.
make: *** [results/nangate45/gcd/base/1_1_yosys.v] Error 1

How can I solve this? Thanks!

vijayank88, 2022-09-16:
@sharkoo7 
Have you verified installation as per this [instructions](https://openroad.readthedocs.io/en/latest/user/BuildWithDocker.html)?

sharkoo7, 2022-09-16:
Yes,I did. Here is what I did:

bash-4.2$ source ./setup_env.sh 
OPENROAD: /OpenROAD-flow-scripts/tools/OpenROAD
bash-4.2$ yosys -help

Usage: yosys [options] [<infile> [..]]

    -Q
        suppress printing of banner (copyright, disclaimer, version)

    -T
        suppress printing of footer (log hash, version, timing statistics)

    -q
        quiet operation. only write warnings and error messages to console
        use this option twice to also quiet warning messages

    -v <level>
        print log headers up to level <level> to the console. (this
        implies -q for everything except the 'End of script.' message.)

    -t
        annotate all log messages with a time stamp

    -d
        print more detailed timing stats at exit

    -l logfile
        write log messages to the specified file

    -L logfile
        like -l but open log file in line buffered mode

    -o outfile
        write the design to the specified file on exit

    -b backend
        use this backend for the output file specified on the command line

    -f frontend
        use the specified frontend for the input files on the command line

    -H
        print the command list

    -h command
        print the help message for the specified command

    -s scriptfile
        execute the commands in the script file

    -c tcl_scriptfile
        execute the commands in the tcl script file (see 'help tcl' for details)

    -p command
        execute the commands

    -m module_file
        load the specified module (aka plugin)

    -X
        enable tracing of core data structure changes. for debugging

    -M
        will slightly randomize allocated pointer addresses. for debugging

    -A
        will call abort() at the end of the script. for debugging

[... 189 lines cut ...]
4.21. Executing OPT pass (performing simple optimizations).
4.21.1. Executing OPT_EXPR pass (perform const folding).
4.21.2. Executing OPT_MERGE pass (detect identical cells).
4.21.3. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
4.21.4. Executing OPT_REDUCE pass (consolidate $*mux and $reduce_* inputs).
4.21.5. Executing OPT_MERGE pass (detect identical cells).
4.21.6. Executing OPT_SHARE pass.
4.21.7. Executing OPT_DFF pass (perform DFF optimizations).
4.21.8. Executing OPT_CLEAN pass (remove unused cells and wires).
4.21.9. Executing OPT_EXPR pass (perform const folding).
4.21.10. Finished OPT passes. (There is nothing left to do.)
4.22. Executing TECHMAP pass (map to technology primitives).
4.22.1. Executing Verilog-2005 frontend: /OpenROAD-flow-scripts/tools/install/yosys/bin/../share/yosys/techmap.v
4.22.2. Continuing TECHMAP pass.
4.23. Executing OPT pass (performing simple optimizations).
4.23.1. Executing OPT_EXPR pass (perform const folding).
4.23.2. Executing OPT_MERGE pass (detect identical cells).
4.23.3. Executing OPT_DFF pass (perform DFF optimizations).
4.23.4. Executing OPT_CLEAN pass (remove unused cells and wires).
4.23.5. Finished fast OPT passes.
4.24. Executing ABC pass (technology mapping using ABC).
4.24.1. Extracting gate netlist of module `\gcd' to `<abc-temp-dir>/input.blif'..
ERROR: ABC: execution of command "/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc -s -f /tmp/yosys-abc-kpTFkt/abc.script 2>&1" failed: return code 127.
Elapsed time: 0:01.30[h:]min:sec. CPU time: user 0.81 sys 0.15 (74%). Peak memory: 32732KB.
make: *** [results/nangate45/gcd/base/1_1_yosys.v] Error 1

vijayank88, 2022-09-17:
Share following details:
`docker --version` 
 OS type:

sharkoo7, 2022-09-17:
I'm using Docker version 20.10.17 and Ubuntu 21.10.
I'm not sure if it's related to yosys-abc,cuz:
```
bash-4.2$ ldd tools/install/yosys/bin/yosys-abc 
tools/install/yosys/bin/yosys-abc: /lib64/libm.so.6: version `GLIBC_2.29' not found (required by tools/install/yosys/bin/yosys-abc)
tools/install/yosys/bin/yosys-abc: /lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by tools/install/yosys/bin/yosys-abc)
tools/install/yosys/bin/yosys-abc: /lib64/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by tools/install/yosys/bin/yosys-abc)
tools/install/yosys/bin/yosys-abc: /lib64/libc.so.6: version `GLIBC_2.34' not found (required by tools/install/yosys/bin/yosys-abc)
	linux-vdso.so.1 =>  (0x00007fff458e4000)
	libreadline.so.8 => not found
	libm.so.6 => /lib64/libm.so.6 (0x00007f72f1c6a000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f72f1962000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f72f174c000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f72f137e000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f72f314b000)
```
How can I fix it?

vijayank88, 2022-09-17:
https://github.com/The-OpenROAD-Project/OpenROAD#install-dependencies Have installed these dependencies?

sharkoo7, 2022-09-17:
Yes I did. But `Package 'libtcl' has no installation candidate` so I changed it into libtcl8.6
I tried to run here DependencyInstaller.sh again.Here is the results: 
```
jkl@ubuntu:~/Desktop/OpenROAD-flow-scripts/tools/OpenROAD$ sudo ./etc/DependencyInstaller.sh -run
Hit:1 http://mirrors.aliyun.com/ubuntu bionic InRelease
Hit:2 http://mirrors.aliyun.com/ubuntu bionic-security InRelease                                 
Hit:3 http://mirrors.aliyun.com/ubuntu bionic-updates InRelease                                  
Hit:4 https://download.docker.com/linux/ubuntu bionic InRelease                                  
Hit:5 http://packages.microsoft.com/repos/code stable InRelease                                  
Hit:6 http://mirrors.aliyun.com/ubuntu bionic-proposed InRelease
Hit:7 http://mirrors.aliyun.com/ubuntu bionic-backports InRelease
Reading package lists... Done                         
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
tzdata is already the newest version (2022c-0ubuntu0.18.04.0).
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libpython3.8 is already the newest version (3.8.0-3ubuntu1~18.04.2).
binutils is already the newest version (2.37-7ubuntu1).
libgomp1 is already the newest version (11.2.0-7ubuntu2).
libtcl8.6 is already the newest version (8.6.11+dfsg-1).
qt5-default is already the newest version (5.99.99+fake-13ubuntu37).
qt5-image-formats-plugins is already the newest version (5.15.2-2).
tcl-tclreadline is already the newest version (2.3.8-1).
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
jkl@ubuntu:~/Desktop/OpenROAD-flow-scripts/tools/OpenROAD$ sudo ./etc/DependencyInstaller.sh -dev
Hit:1 http://mirrors.aliyun.com/ubuntu bionic InRelease
Hit:2 http://mirrors.aliyun.com/ubuntu bionic-security InRelease                             
Hit:3 http://mirrors.aliyun.com/ubuntu bionic-updates InRelease                              
Hit:4 https://download.docker.com/linux/ubuntu bionic InRelease                             
Hit:5 http://mirrors.aliyun.com/ubuntu bionic-proposed InRelease
Hit:6 http://mirrors.aliyun.com/ubuntu bionic-backports InRelease                           
Hit:7 http://packages.microsoft.com/repos/code stable InRelease                             
Reading package lists... Done                            
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
tzdata is already the newest version (2022c-0ubuntu0.18.04.0).
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libpython3.8 is already the newest version (3.8.0-3ubuntu1~18.04.2).
binutils is already the newest version (2.37-7ubuntu1).
libgomp1 is already the newest version (11.2.0-7ubuntu2).
libtcl8.6 is already the newest version (8.6.11+dfsg-1).
qt5-default is already the newest version (5.99.99+fake-13ubuntu37).
qt5-image-formats-plugins is already the newest version (5.15.2-2).
tcl-tclreadline is already the newest version (2.3.8-1).
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
Hit:1 http://mirrors.aliyun.com/ubuntu bionic InRelease
Hit:2 http://mirrors.aliyun.com/ubuntu bionic-security InRelease                                 
Hit:3 http://mirrors.aliyun.com/ubuntu bionic-updates InRelease                                  
Hit:4 http://mirrors.aliyun.com/ubuntu bionic-proposed InRelease                                 
Hit:5 http://mirrors.aliyun.com/ubuntu bionic-backports InRelease                                
Hit:6 https://download.docker.com/linux/ubuntu bionic InRelease                                  
Hit:7 http://packages.microsoft.com/repos/code stable InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
tzdata is already the newest version (2022c-0ubuntu0.18.04.0).
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
automake is already the newest version (1:1.16.4-2).
autotools-dev is already the newest version (20180224.1+nmu1).
bison is already the newest version (2:3.7.6+dfsg-1build1).
clang is already the newest version (1:13.0-53~exp1).
flex is already the newest version (2.6.4-8build1).
g++ is already the newest version (4:11.2.0-1ubuntu1).
gcc is already the newest version (4:11.2.0-1ubuntu1).
git is already the newest version (1:2.32.0-1ubuntu1.3).
lcov is already the newest version (1.14-2).
libpcre3-dev is already the newest version (2:8.39-13ubuntu0.21.10.1).
libreadline-dev is already the newest version (8.1-2).
python3-dev is already the newest version (3.9.4-1build1).
tcl-dev is already the newest version (8.6.11+1build1).
tcllib is already the newest version (1.20+dfsg-1).
wget is already the newest version (1.21-1ubuntu3).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-2ubuntu7.1).
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
CMake already installed.
Swig already installed.
Boost already installed.
Eigen already installed.
CUSP already installed.
Lemon already installed.
spdlog already installed.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.

```

vijayank88, 2022-09-17:
Its working fine with Ubuntu 20.04. 
@vvbandeira any help?

sharkoo7, 2022-09-18:
Thanks a lot for helping me. I deleted the whole directory,cloned and built all over again. It worked fine this time!

closed 2022-09-18 by sharkoo7