The-OpenROAD-Project/OpenROAD-flow-scripts #597: Add Antenna information back to nangate45

rovinski, 2022-09-26:
**Is your feature request related to a problem? Please describe.**
Nangate45 has antenna information in the [original LEF](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/bb1bc7d63c268417868c994396f8702a0460b23c/flow/platforms/nangate45/lef/NangateOpenCellLibrary.macro.lef) but it was stripped out when converting the LEF pin shapes from `POLYGON` to `RECT` in [NangateOpenCellLibrary.macro.mod.lef](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/bb1bc7d63c268417868c994396f8702a0460b23c/flow/platforms/nangate45/lef/NangateOpenCellLibrary.macro.mod.lef).

**Describe the solution you'd like**
Add the antenna information back to the LEF.

**Describe alternatives you've considered**
N/A

rovinski, 2022-09-26:
@maliberty Does DRT still have a limitation that it can only read `RECT` pins? Or has the odb integration resolved that? If so, we can delete the modified LEF and go back to using the original with `POLYGON` pins.

maliberty, 2022-09-26:
I think it will work - can you give it a try?

rovinski, 2026-10-09:
Still valid AFAICT