The-OpenROAD-Project/OpenROAD-flow-scripts #599: cannot open openroad gui

sharkoo7, 2022-09-27:
After running designs/sky130hd/chameleon, I wanted to view the result with openroad gui. But` make gui_final DESIGN_CONFIG=./designs/sky130hd/chameleon/config.mk` stucked at` loading spef`.  But make gui_floorplan `DESIGN_CONFIG=./designs/sky130hd/chameleon/config.mk` didn't work as well. I wonder what's the problem?

rovinski, 2022-09-27:
Please use the bug report template. It's there for a reason. Please state:
1. Your OpenROAD-flow-scrips version (git hash)
2. Your OS and version
3. How you are running it (natively or Docker)
4. Attach the full log

sharkoo7, 2022-09-30:
1. c086c4689266038b03334d1198d1e0f0369df87d
2. Ubuntu 22.04.1 LTS
3. I am running it in docker
4. [log.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9681776/log.zip)
5.after completing the flow,` make gui_final DESIGN_CONFIG=./designs/sky130hd/chameleon/config.mk` stucked at` loading spef`.
[image]

maliberty, 2022-10-02:
Would you try setting GUI_NO_TIMING=1 and see if the problem persists or not?  That will skip loading the spef.

alybruin, 2022-11-17:
Hi @sharkoo7 .  I'm currently having the same issue with "make gui_final" stuck at "loading spef".  Did you ever resolve your issue and if yes, could you share your solution? Thank you very much in advance.

maliberty, 2022-11-21:
@alybruin can you try my suggestion and report if it works?

alybruin, 2022-11-21:
Hi @maliberty , yes I have tried your suggestion but still same issue with stuck at "loading spef".  In gui.tcl script, the GUI_NO_TIMING param is used to determine design_stage 1 - 6.  If design-stage = 6, it performs "Loading spef".  Else, it performs "Estimate parasitic".  Even when I commented out both loading spef and estimating parasitic, the GUI is still stuck and not opening.

maliberty, 2022-11-21:
GUI_NO_TIMING will skip the loading or estimating of any parasitics.  If you have set it you should not get a "Loading spef" message at all.

So the problem is not related to parasitics.  How large is your design?

maliberty, 2022-11-21:
Can you provide your .odb file?

maliberty, 2022-11-21:
and the version of OR you are using (openroad -version)

alybruin, 2022-11-21:
Hi @maliberty .  Sorry for not being clear earlier.  After setting GUI_NO_TIMING, I do not get "Loading spef" message anymore.  However, it's still not opening GUI, just stuck there.

alybruin, 2022-11-21:
Hi @maliberty .  I'm running the example ethmac design using asap7 platform.

maliberty, 2022-11-21:
What version of OR?
What platform?
Are you using docker?

alybruin, 2022-11-21:
@maliberty .  I'm running CentOS7 and using Docker.  As for OR version, I'm still trying to figure that out.  I'm using OR flow scripts, not OpenLane.  The way I tried to open GUI is in Docker, after make successful, I type "make gui-final".  Also, I can navigate to OpenROAD-flow-scripts/tools/install/OpenROAD/bin  and typing "./openroad -version" returns GITDIR-NOTFOUND and "./openroad -gui" returns Could not connect to any X display.

maliberty, 2022-11-21:
I'm guessing this is a docker issue and not an OR issue.  In OL there is much discussion in https://github.com/The-OpenROAD-Project/OpenLane/issues/1149.  Would you try running docker with --privileged and see if that fixes it.

I can open that design in the gui with timing in <1m outside of docker.

alybruin, 2022-11-21:
Hi @maliberty .  I did try adding that "DOCKER_OPTIONS += --privileged" but same issue.  Thank you for confirming that you can open GUI outside of Docker.  That may be a clue.

maliberty, 2022-11-21:
You said that you are using ORFS so DOCKER_OPTIONS won't do anything as it is an OpenLane option.  How are you invoking docker in orfs?  You'll need to add the option there directly.

alybruin, 2022-11-21:
To invoke Docker, I typed      docker run --privileged -it -u $(id -U ${1000}):$(id -g ${1000}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro openroad/flow-scripts    With or without --privileged, same issue.

maliberty, 2022-11-21:
https://github.com/The-OpenROAD-Project/OpenLane/blob/ff85171cfb10dc44911db7a62cc2d86f03807756/Makefile#L34 shows all the options that OL uses.  I suspect you need to mount /tmp/.X11-unix too

maliberty, 2022-11-30:
Did this work for you?

alybruin, 2022-12-02:
Hi Matt.  I tried it while back when you first suggested but same issue.  So it didn't help.  I have put that issue aside for now and will come back to it in the future.  Thanks for all your help.

zerzerzerz, 2022-12-15:
Hi, I encountered the same problem. I finally fix it by the following commands.
I am using `Docker` to build OpenROAD in `Ubuntu-18.04`.
When constructing the container from image, run these following commands. Because it needs GUI, so I choose `MobaXterm` via SSH to connect the remote server, which supports `x11-forwarding`. Then enter these commands to construct an container from the built image. If I specify the username and group when running the container, it failed with `cannot connect to display :10.0`. Maybe I need to configure more settings, but this method works for me. So I just run the container as `root`.
To set the `-e DISPLAY=:10.0`, I execute `echo $DISPLAY` in the terminal in remote server with x11-forwarding. The output is `localhost:10.0`, so I set `-e DISPLAY=:10.0`.
* Connect to remote server with MobaXterm (supporting x11-forwarding) via SSH
```bash
ssh -p <port> <username>@<ip_address>
```
* Check display device in remote server, and my output is `localhost:10.0`
```bash
echo $DISPLAY
```
* Construct and run the container from image, mount coresponding dirs with `-v` and set display with `-e DISPLAY=<YOUR_DISPLAY>`
```bash
docker run \
  -it \
  --privileged \
  -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro \
  -v /etc/localtime:/etc/localtime \
  -v $HOME/.Xauthority:/root/.Xauthority \
  -e DISPLAY=:10.0 \
  --net=host \
  openroad/flow-scripts
```
* After executing these commands, just follow the official instructions to run
```bash
source setup_env.sh
openroad -gui
``` 
Finally I can use the GUI.
Hope this works for you~

closed 2022-12-15 by maliberty

Amik-TJ, 2024-10-18:
> Hi, I encountered the same problem. I finally fix it by the following commands. I am using `Docker` to build OpenROAD in `Ubuntu-18.04`. When constructing the container from image, run these following commands. Because it needs GUI, so I choose `MobaXterm` via SSH to connect the remote server, which supports `x11-forwarding`. Then enter these commands to construct an container from the built image. If I specify the username and group when running the container, it failed with `cannot connect to display :10.0`. Maybe I need to configure more settings, but this method works for me. So I just run the container as `root`. To set the `-e DISPLAY=:10.0`, I execute `echo $DISPLAY` in the terminal in remote server with x11-forwarding. The output is `localhost:10.0`, so I set `-e DISPLAY=:10.0`.
> 
> * Connect to remote server with MobaXterm (supporting x11-forwarding) via SSH
> 
> ```shell
> ssh -p <port> <username>@<ip_address>
> ```
> 
> * Check display device in remote server, and my output is `localhost:10.0`
> 
> ```shell
> echo $DISPLAY
> ```
> 
> * Construct and run the container from image, mount coresponding dirs with `-v` and set display with `-e DISPLAY=<YOUR_DISPLAY>`
> 
> ```shell
> docker run \
>   -it \
>   --privileged \
>   -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro \
>   -v /etc/localtime:/etc/localtime \
>   -v $HOME/.Xauthority:/root/.Xauthority \
>   -e DISPLAY=:10.0 \
>   --net=host \
>   openroad/flow-scripts
> ```
> 
> * After executing these commands, just follow the official instructions to run
> 
> ```shell
> source setup_env.sh
> openroad -gui
> ```
> 
> Finally I can use the GUI. Hope this works for you~

This worked for me. Thanks a lot

maliberty, 2024-10-18:
You can use https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/util/docker_shell