The-OpenROAD-Project/OpenROAD-flow-scripts #605: Flow exiting in CTS stage with an OverflowError

stevenmburns, 2022-10-07:
**Describe the bug**
I have a testcase that fails in the CTS stage returning this error:
```
==========================================================================
cts pre-repair report_tns
--------------------------------------------------------------------------
tns 0.00

==========================================================================
cts pre-repair report_wns
--------------------------------------------------------------------------
wns 0.00

==========================================================================
cts pre-repair report_worst_slack
--------------------------------------------------------------------------
worst slack 799.50

==========================================================================
cts pre-repair report_clock_skew
--------------------------------------------------------------------------
Clock main_clk
No launch/capture paths found.

Error: cts.tcl, 44 OverflowError in method 'metric_float', argument 2 of type 'float'
Elapsed time: 0:02.74[h:]min:sec. CPU time: user 2.70 sys 0.04 (99%). Peak memory: 100904KB.
make: *** [results/nangate45/priority_mux_256/base/4_1_cts.odb] Error 1
```
**Expected behavior**
This should run through the flow and produce a layout, or give a description of the input errors.

**Screenshots**
N/A

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: Centos 7 in an Apptainer container
 - OpenROAD-flow commit: 0317619cb67f1c77437a0aad479e4902f7cffa1b

**File Uploads**
[bug.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/9730578/bug.tar.gz)
```
drwxr-x--- smburns/scl       0 2022-10-06 19:41 designs/src/priority_mux_256/
-rw-r----- smburns/scl   98233 2022-10-06 19:41 designs/src/priority_mux_256/priority_mux_256.v
drwxr-x--- smburns/scl       0 2022-10-06 20:50 designs/nangate45/priority_mux_256/
-rw-r----- smburns/scl     432 2022-10-06 20:47 designs/nangate45/priority_mux_256/constraint.sdc
-rw-r----- smburns/scl   12683 2022-10-06 20:45 designs/nangate45/priority_mux_256/metadata-base-ok.json
-rw-r----- smburns/scl    2323 2022-10-06 20:45 designs/nangate45/priority_mux_256/rules-base.json
-rw-r----- smburns/scl     480 2022-10-06 20:50 designs/nangate45/priority_mux_256/config.mk
```

**Additional context**
The bug was discovered on a different platform (intel16) and converted to nangate45 for debug.
Use to reproduce, use this config after incorporating the design files above.
```
DESIGN_CONFIG=./designs/nangate45/priority_mux_256/config.mk
```

maliberty, 2022-10-07:
I expect this is the same as https://github.com/The-OpenROAD-Project/OpenROAD/issues/2268

The fix was merged but had to be reverted due to other issues.  It is planned to go in again soon.

maliberty, 2022-10-07:
You can work around this by disabling clock skew reporting in flow/scripts/report_metrics.tcl

closed 2022-10-07 by stevenmburns