The-OpenROAD-Project/OpenROAD-flow-scripts #608: Error in floorplan tcl on asap7 gcd regression test

tspyrou, 2022-10-11:
git clone --recursive git@github.com:The-OpenROAD-Project/OpenROAD-flow-scripts.git
cd OpenROAD-flow-scripts/
 ./build_openroad.sh --local --latest --nice
cd flow
make DESIGN_CONFIG=./designs/asap7/gcd/config.mk

Error: floorplan.tcl, 101 invalid command name "report_power_metric"
Command exited with non-zero status 1
Elapsed time: 0:01.37[h:]min:sec. CPU time: user 1.27 sys 0.09 (99%). Peak memory: 184724KB.
make: *** [Makefile:414: results/asap7/gcd/base/2_1_floorplan.odb] Error 1

tspyrou, 2022-10-11:
This is the problem @ravi-varadarajan 

tspyrou@DESKTOP-JDHB6MS:~/OpenROAD-flow-scripts/flow$ git diff
diff --git a/flow/scripts/report_metrics.tcl b/flow/scripts/report_metrics.tcl
index b885d9f3..cde7b71a 100644
--- a/flow/scripts/report_metrics.tcl
+++ b/flow/scripts/report_metrics.tcl
@@ -154,12 +154,12 @@ proc report_metrics { when {include_erc true} {include_clock_skew true} } {
     foreach corner $::env(CORNERS) {
       puts "Corner: $corner"
       report_power -corner $corner
-      report_power_metric -corner $corner
+#      report_power_metric -corner $corner
     }
     unset corner
   } else {
     report_power
-    report_power_metric
+#    report_power_metric
   }

   puts "\n=========================================================================="

maliberty, 2022-10-11:
Are you using the head of OR?  report_power_metric is defined in Metrics.tcl and is available in OR.

maliberty, 2022-10-11:
@tspyrou I did your sequence of steps without any error.

maliberty, 2022-10-11:
It also passes in the CI https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Nightly-Public/1687/execution/node/434/log/?consoleFull

closed 2022-10-11 by maliberty

ravi-varadarajan, 2022-10-12:
This was reported by UCSC -- Jignesh Shah.    SO the solution for them is ensure that they are using head of OR.  I also suspected it was OR version issue.