The-OpenROAD-Project/OpenROAD-flow-scripts #612: ASAP7 M2 has very large pitch

antonblanchard, 2022-10-13:
I'm debugging an issue where global routing is routing many connections between adjacent cells all the way through M4. I'm testing with `designs/asap7/aes`. In all of these cases detailed routing was able to satisfy it using M2.

Even after changing `set_global_routing_layer_adjustment`:

```
-  set_global_routing_layer_adjustment $env(MIN_ROUTING_LAYER)-$env(MAX_ROUTING_LAYER) 0.5
+  set_global_routing_layer_adjustment $env(MIN_ROUTING_LAYER)-$env(MAX_ROUTING_LAYER) 0.2
```

I still saw many cases of this issue. M2 capacity is tiny compared to layers above it:

```
[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
M1         Vertical            0             0          0.00%
M2         Horizontal      31752         16000          49.61%
M3         Vertical       238140        205185          13.84%
M4         Horizontal     174636        143375          17.90%
M5         Vertical       174636        141239          19.12%
M6         Horizontal      79380         60650          23.60%
M7         Vertical        79380         64500          18.75%
---------------------------------------------------------------
```

If I look back further, M2 has a much bigger pitch than M4, which doesn't make sense to me:

```
[INFO GRT-0088] Layer M2      Track-Pitch = 0.2700  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M3      Track-Pitch = 0.0360  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M4      Track-Pitch = 0.0480  line-2-Via Pitch: 0.0480
```

It looks like we are pulling this pitch from odb, which made me look at floor planning, specifically the asap7 specific make_tracks. This fixes the M2 pitch issue, but we do end up with 1 DRC issue:

```
-make_tracks M2  -x_offset [expr 0.009 * $multiplier] -x_pitch [expr 0.036 * $multiplier] -y_offset [expr (0.045 - 0.000) * $multiplier] -y_pitch [expr 0.270 * $multiplier]
-make_tracks M2  -x_offset [expr 0.009 * $multiplier] -x_pitch [expr 0.036 * $multiplier] -y_offset [expr (0.081 - 0.000) * $multiplier] -y_pitch [expr 0.270 * $multiplier]
-make_tracks M2  -x_offset [expr 0.009 * $multiplier] -x_pitch [expr 0.036 * $multiplier] -y_offset [expr (0.117 - 0.000) * $multiplier] -y_pitch [expr 0.270 * $multiplier]
-make_tracks M2  -x_offset [expr 0.009 * $multiplier] -x_pitch [expr 0.036 * $multiplier] -y_offset [expr (0.153 - 0.000) * $multiplier] -y_pitch [expr 0.270 * $multiplier]
-make_tracks M2  -x_offset [expr 0.009 * $multiplier] -x_pitch [expr 0.036 * $multiplier] -y_offset [expr (0.189 - 0.000) * $multiplier] -y_pitch [expr 0.270 * $multiplier]
-make_tracks M2  -x_offset [expr 0.009 * $multiplier] -x_pitch [expr 0.036 * $multiplier] -y_offset [expr (0.225 - 0.000) * $multiplier] -y_pitch [expr 0.270 * $multiplier]
-make_tracks M2  -x_offset [expr 0.009 * $multiplier] -x_pitch [expr 0.036 * $multiplier] -y_offset [expr (0.270 - 0.000) * $multiplier] -y_pitch [expr 0.270 * $multiplier]
+make_tracks M2  -x_offset [expr 0.009 * $multiplier] -x_pitch [expr 0.036 * $multiplier] -y_offset [expr 0.009 * $multiplier] -y_pitch [expr 0.036 * $multiplier]
```

I'm not sure if the problem is in how ASAP7 is calling make_tracks, or how odb is calculating the pitch.

maliberty, 2022-10-13:
I don't understand you command about pitch - I see m2 having a smaller pitch than m4.   In any case the pitch isn't very useful as we have a custom track pattern for m2 as you note.  m2 does have power rails which will reduce its capacity relative to higher layers but I'll still take a look at what's going on.

maliberty, 2022-10-13:
There is a real problem in that GlobalRouter::initRoutingTracks is only getting one grid pattern back 
```
track_grid->getGridPatternY(0, ...
```
and not all of them.  There is a strong assumption of a single track pattern which is not right here.  I'll have to think about how best to fix this for a bit.

maliberty, 2022-10-13:
Fixed with the above PR

closed 2022-10-13 by maliberty