The-OpenROAD-Project/OpenROAD-flow-scripts #663: OpenROAD-flow-scripts build not succesfsul

lulu9312, 2022-11-30:
### Subject

[Build] for compilation-related issues.

### Describe the bug

I am updating OpenROAD-flow-scripts by using command `./build_openroad.sh --local --latest` but the build is not successful and gives the below error.
As @vijayank88 suggested in the slack thread  I also tried using commands 
`sudo git clean -xdf .tools`
`sudo git submodule foreach --recursive git clean -xdf`
but stills the error persists.

### Expected Behavior

Build Complete.

### Environment

```shell
Git commit: 0769b42c9bb1b34f7a899ed02551c441d0df4eaf
kernel: Linux 5.15.0-53-generic
os: Ubuntu 20.04.5 LTS (Focal Fossa)
cmake version 3.24.2
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
clang version 10.0.0-4ubuntu1
```

### To Reproduce

This is a build issue.

### Screenshots

[image]

vvbandeira, 2022-11-30:
> sudo git clean -xdf .tools

@lulu9312  Did you run this exactly? There's a typo above (extra period); it should be: 
```
sudo git clean -xdf tools
```

Or you could use the following, which will do the commands automatically (and also save the log):
```
./build_openroad.sh --local --latest --clean-force &| tee ../build.log
```

If the problem still persists, please upload a full log of the build. The build happens in parallel, so only the last couple of lines from the screenshot do not show what actually caused the error.

lulu9312, 2022-11-30:
@vvbandeira There was typo while writing the earlier command , I had used sudo git clean -xdf ./tools. 
Now I used this command `./build_openroad.sh --local --latest --clean-force 2>&1 | tee ../build.log` and same error persists. Also where does the build.log file get generated could not find it.

rovinski, 2022-11-30:
@lulu9312 please also do not take screenshots of the log, you can simply paste the text into the `Relevant log output` field. This makes it easier to read and also allows the log to be indexable in web searches for others who have similar problems.

Beyond that, it is important to include the full log for build errors. The build script is multithreaded so the last text you see does not mean that is where the error came from. The error could have occurred much earlier in the log.

@vvbandeira perhaps `build_openroad.sh` should create a log by default so that it can be uploaded?

lulu9312, 2022-11-30:
@rovinski Got it.I will keep that in mind. 
@vvbandeira Found the log file. Here's the build.log
[build.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/10124898/build.log)

rovinski, 2022-11-30:
Maybe the log file is corrupted but something looks very wrong with your boost headers:
```
In file included from /usr/local/include/boost/fusion/adapted/struct/detail/adapt_base.hpp:18,
                 from /usr/local/include/boost/fusion/adapted/struct/adapt_struct.hpp:23,
                 from /usr/local/include/boost/fusion/include/adapt_struct.hpp:12,
                 from /home/pranav/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lefin/boostParser.h:35,
                 from /home/pranav/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/lefin/lefTechLayerSpacingEolParser.cpp:31:
/usr/local/include/boost/fusion/adapted/struct/detail/adapt_is_tpl.hpp:1:2: error: stray â€˜\10â€™ in program
    1 |   )$‚THDÌ  ³Žß~ À³-h:Mß´&€ øîM`<d ³ðá!õ„ß.  Àxør¡e8è= ³ (x8ës‡Ž‡ãáŠ3º6 <Á <€èûs‘Èò0’3Îç0åÊ)ñÁŽÎ›HÀdïMþ#}Â×D(¬Ph½­ÿOƒ¨hJÌ´hÏƒðºç9g@`‡<„R|9˜® ³h|l ‚b|þ=D \& “ÿ–Û»È| hÎxŽ@õƒh|ÂË½wŽø  eô¶…  €8©·8<%: ³Ð€"†ShÏ!pà×#hÏ¬°,€T/hHÊó h8RJ’„'(ëOJTÊI&tJ‘ˆB±35Î,¥”1J;sNÓŒ'*óRJuN)Mh”NÉ;R¹˜3¢5Ád2¥œ2pˆ  œÚ·°(    ú  pÜà ú h¥ÿÖ¢ÔOaü·À¢ õ    ¤‹q(Ð ¨•Õ @¢ »Z Ø ÎÉ™l  p–LrÞÚ ƒOõPèèP´ ³ à‡?Ü¿{ZI°†´¹]øá÷ïž ÿA€5@‚ÍTkp  À.õ€5Â% :O Öàª†  \™³
      |  ^
```
These should be header files not binary files. Try uninstalling and reinstalling Boost maybe?

lulu9312, 2022-12-01:
@rovinski  @vvbandeira The issue is now resolved. I reinstalled Boost.

closed 2022-12-01 by vvbandeira