The-OpenROAD-Project/OpenROAD-flow-scripts #664: Create build_openroad.sh log automatically

rovinski, 2022-11-30:
**Is your feature request related to a problem? Please describe.**
Create a build log for `build_openroad.sh` automatically so that users don't have to follow extra steps when submitting build issues.

**Describe the solution you'd like**
See above

**Describe alternatives you've considered**
N/A

**Additional context**
We could use a solution similar to the one [here](https://stackoverflow.com/questions/3173131/redirect-copy-of-stdout-to-log-file-from-within-bash-script-itself).

rovinski, 2022-11-30:
I can work on this when I get a chance or someone else can take over.

vvbandeira, 2022-12-01:
@habibayassin can you propose a PR?

habibayassin, 2022-12-01:
@vvbandeira yes sure

closed 2022-12-07 by vvbandeira