The-OpenROAD-Project/OpenROAD-flow-scripts #671: Compilation with Docker broken?

davideschiavone, 2022-12-05:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Dear all,

I just followed the steps here 

https://openroad.readthedocs.io/en/latest/user/BuildWithDocker.html

and when launching 

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh

```

I get

```
[ 99%] Building frontends/verilog/verilog_lexer.o
frontends/verilog/verilog_lexer.l: In function 'int frontend_verilog_yylex(FRONTEND_VERILOG_YYSTYPE*, FRONTEND_VERILOG_YYLTYPE*)':
frontends/verilog/verilog_lexer.cc:199:36: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                 for ( yyl = n; yyl < frontend_verilog_yyleng; ++yyl )\
                                ~~~~^~~
frontends/verilog/verilog_lexer.cc:210:9: note: in expansion of macro 'YY_LESS_LINENO'
         YY_LESS_LINENO(yyless_macro_arg);\
         ^~~~~~~~~~~~~~
frontends/verilog/verilog_lexer.l:413:3: note: in expansion of macro 'yyless'
   yyless(len);
   ^~~~~~
[100%] Building yosys
fatal: unable to access 'https://github.com/berkeley-abc/abc/': Could not resolve host: github.com; Unknown error
+ cd abc
/bin/sh: line 4: cd: abc: No such file or directory
make: *** [Makefile:746: abc/abc-bafd2a7] Error 1
```

Do know know why it happens?

### Expected Behavior

Expecting building going through

### Environment

```shell
OpenROAD-flow-scripts/tools/OpenROAD$ ./etc/Env.sh
Please pull the latest changes and try again, if problem persists file a github issue with the re-producible test case.
```

### To Reproduce

git clone last verion (a58140ed1cbcdca2139a0bc910537791bbf611c8) 

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh

davideschiavone, 2022-12-05:
Also when trying locally, I get the following error on a missing file:

```
OpenROAD-flow-scripts$ ./build_openroad.sh --local
[INFO FLW-0002] Updating git submodules.
[INFO FLW-0001] Using local build method. This will create binaries at 'tools/install' unless overwritten.
[INFO FLW-0017] Compiling Yosys.
make: Entering directory '/whateverfolder/OpenROAD-flow-scripts/tools/yosys'
[  0%] Building kernel/version_bc027b2ca.o
[  0%] Building kernel/driver.o
[  1%] Building kernel/register.o
[  1%] Building kernel/rtlil.o
[  1%] Building kernel/log.o
[  2%] Building kernel/calc.o
[  2%] Building kernel/yosys.o
[  2%] Building kernel/binding.o
[  3%] Building kernel/cellaigs.o
[  3%] Building kernel/celledges.o
[  3%] Building kernel/satgen.o
[  4%] Building kernel/qcsat.o
[  4%] Building kernel/mem.o
[  4%] Building kernel/ffmerge.o
[  5%] Building kernel/ff.o
[  6%] Building libs/bigint/BigUnsigned.o
[  7%] Building libs/bigint/BigUnsignedInABase.o
kernel/driver.cc:24:12: fatal error: readline/readline.h: No such file or directory
   24 | #  include <readline/readline.h>
```

vvbandeira, 2022-12-05:
@davideschiavone 
For the docker build, could you try:
```
./build_openroad.sh --clean
```

For the local installation you will need to run the `DependencyInstaller.sh`
```
tools/OpenROAD/etc/DependencyInstaller.sh
```

davideschiavone, 2022-12-05:
thanks @vvbandeira , I have to try the local installation but the docker one went well.

Would it make sense to add the `--clean` option to the documentation?

vvbandeira, 2022-12-05:
@davideschiavone good to know you were able to compile openroad, and thanks for the suggestion.

@habibayassin can you add a note/section for troubleshooting and suggest the use of the `--clean` argument as a first option?

(closing as complete since the problem was fixed, discussion can continue if needed)

closed 2022-12-05 by vvbandeira

davideschiavone, 2022-12-06:
hi @vvbandeira , the local build is still broken with the same errors.

I did from a fresh clone:

```
sh ./tools/OpenROAD/etc/DependencyInstaller.sh
./build_openroad.sh --local
```
and this is what I got

```
[INFO FLW-0002] Updating git submodules.
[INFO FLW-0001] Using local build method. This will create binaries at 'tools/install' unless overwritten.
[INFO FLW-0017] Compiling Yosys.
make: Entering directory '/home/mynicefolder/OpenROAD-flow-scripts/tools/yosys'
[  0%] Building kernel/version_bc027b2ca.cc
[  0%] Building kernel/driver.o
[  0%] Building techlibs/common/simlib_help.inc
[  0%] Building techlibs/common/simcells_help.inc
[  1%] Building kernel/rtlil.o
[  1%] Building kernel/log.o
[  2%] Building kernel/calc.o
[  2%] Building kernel/yosys.o
[  2%] Building kernel/binding.o
[  3%] Building kernel/cellaigs.o
[  3%] Building kernel/celledges.o
[  3%] Building kernel/satgen.o
[  4%] Building kernel/qcsat.o
[  4%] Building kernel/mem.o
[  4%] Building kernel/ffmerge.o
[  5%] Building kernel/ff.o
kernel/driver.cc:24:12: fatal error: readline/readline.h: No such file or directory
   24 | #  include <readline/readline.h>
      |            ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:705: kernel/driver.o] Error 1
make: *** Waiting for unfinished jobs....
[  5%] Building libs/bigint/BigIntegerAlgorithms.o
kernel/yosys.cc:24:12: fatal error: readline/readline.h: No such file or directory
   24 | #  include <readline/readline.h>
```

davideschiavone, 2022-12-06:
sudo apt-get install libreadline-dev 
solved the issue, would it make sense to add it to the dependencies?

vvbandeira, 2022-12-06:
@davideschiavone it already is, see:

https://github.com/The-OpenROAD-Project/OpenROAD/blob/9eba9fc71a37c60eb3fcc45d4e0f29fc2fc610d6/etc/DependencyInstaller.sh#L160