The-OpenROAD-Project/OpenROAD-flow-scripts #675: Unable to build with docker

zerzerzerz, 2022-12-07:
### Subject

[Documentation] for documentation errors.

### Describe the bug

I try to build OpenROAD with docker in WSL2-Ubuntu-18.04.
I executed these commands
```bash
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh
```
but got
```
#8 592.4 /opt/rh/devtoolset-8/root/usr/bin/ranlib libabc.a
#8 592.6 gmake[2]: Leaving directory '/OpenROAD/build'
#8 592.6 [ 79%] Built target libabc
#8 592.6 gmake[1]: Leaving directory '/OpenROAD/build'
#8 592.6 gmake: *** [Makefile:139: all] Error 2
------
executor failed running [/bin/sh -c ./etc/Build.sh -compiler=${compiler} -threads=${numThreads}]: exit code: 2
```
More detailed error information could be found in `Relevant log output`

### Expected Behavior

Build docker image successfully

### Environment

```shell
-- The CXX compiler identification is GNU 8.4.0
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-5852-g2a9ef5fa1
-- System name: Linux
-- Compiler: GNU 8.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- The C compiler identification is GNU 8.4.0
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
[... 72 lines cut ...]
cd OpenROAD-flow-scripts
./build_openroad.sh
```

### Relevant log output

```shell
#8 590.0 [ 79%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb4Sweep.c.o
#8 590.0 cd /OpenROAD/build/third-party/abc && /opt/rh/devtoolset-8/root/usr/bin/g++  -I/OpenROAD/third-party/abc/src -O3 -DNDEBUG -Wall -Wno-array-bounds -Wno-nonnull -Wno-maybe-uninitialized -Wno-format-overflow -Wno-unused-variable -Wno-unused-function -Wno-write-strings -Wno-sign-compare -Wno-deprecated -Wno-register -Wno-format -DLIN64 -DSIZEOF_VOID_P=8 -DSIZEOF_LONG=8 -DSIZEOF_INT=4 -DABC_NAMESPACE=abc -fpermissive -DABC_USE_CUDD=1 -DABC_USE_PTHREADS -Wno-unused-but-set-variable -std=c++17 -MD -MT third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb4Sweep.c.o -MF CMakeFiles/libabc.dir/src/bdd/llb/llb4Sweep.c.o.d -o CMakeFiles/libabc.dir/src/bdd/llb/llb4Sweep.c.o -c /OpenROAD/third-party/abc/src/bdd/llb/llb4Sweep.c
#8 591.9 [ 79%] Linking CXX static library libabc.a
#8 591.9 cd /OpenROAD/build/third-party/abc && /usr/local/bin/cmake -P CMakeFiles/libabc.dir/cmake_clean_target.cmake
#8 591.9 cd /OpenROAD/build/third-party/abc && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/libabc.dir/link.txt --verbose=1
#8 591.9 /opt/rh/devtoolset-8/root/usr/bin/ar qc libabc.a CMakeFiles/libabc.dir/src/base/abc/abcAig.c.o CMakeFiles/libabc.dir/src/base/abc/abcBarBuf.c.o CMakeFiles/libabc.dir/src/base/abc/abcBlifMv.c.o CMakeFiles/libabc.dir/src/base/abc/abcCheck.c.o CMakeFiles/libabc.dir/src/base/abc/abcDfs.c.o CMakeFiles/libabc.dir/src/base/abc/abcFanio.c.o CMakeFiles/libabc.dir/src/base/abc/abcFanOrder.c.o CMakeFiles/libabc.dir/src/base/abc/abcFunc.c.o CMakeFiles/libabc.dir/src/base/abc/abcHie.c.o CMakeFiles/libabc.dir/src/base/abc/abcHieCec.c.o CMakeFiles/libabc.dir/src/base/abc/abcHieGia.c.o CMakeFiles/libabc.dir/src/base/abc/abcHieNew.c.o CMakeFiles/libabc.dir/src/base/abc/abcLatch.c.o CMakeFiles/libabc.dir/src/base/abc/abcLib.c.o CMakeFiles/libabc.dir/src/base/abc/abcMinBase.c.o CMakeFiles/libabc.dir/src/base/abc/abcNames.c.o CMakeFiles/libabc.dir/src/base/abc/abcNetlist.c.o CMakeFiles/libabc.dir/src/base/abc/abcNtk.c.o CMakeFiles/libabc.dir/src/base/abc/abcObj.c.o CMakeFiles/libabc.dir/src/base/abc/abcRefs.c.o CMakeFiles/libabc.dir/src/base/abc/abcShow.c.o CMakeFiles/libabc.dir/src/base/abc/abcSop.c.o CMakeFiles/libabc.dir/src/base/abc/abcUtil.c.o CMakeFiles/libabc.dir/src/base/abci/abc.c.o CMakeFiles/libabc.dir/src/base/abci/abcAttach.c.o CMakeFiles/libabc.dir/src/base/abci/abcAuto.c.o CMakeFiles/libabc.dir/src/base/abci/abcBalance.c.o CMakeFiles/libabc.dir/src/base/abci/abcBidec.c.o CMakeFiles/libabc.dir/src/base/abci/abcBm.c.o CMakeFiles/libabc.dir/src/base/abci/abcBmc.c.o CMakeFiles/libabc.dir/src/base/abci/abcCas.c.o CMakeFiles/libabc.dir/src/base/abci/abcCascade.c.o CMakeFiles/libabc.dir/src/base/abci/abcCollapse.c.o CMakeFiles/libabc.dir/src/base/abci/abcCut.c.o CMakeFiles/libabc.dir/src/base/abci/abcDar.c.o CMakeFiles/libabc.dir/src/base/abci/abcDebug.c.o CMakeFiles/libabc.dir/src/base/abci/abcDec.c.o CMakeFiles/libabc.dir/src/base/abci/abcDetect.c.o CMakeFiles/libabc.dir/src/base/abci/abcDress.c.o CMakeFiles/libabc.dir/src/base/abci/abcDress2.c.o CMakeFiles/libabc.dir/src/base/abci/abcDress3.c.o CMakeFiles/libabc.dir/src/base/abci/abcDsd.c.o CMakeFiles/libabc.dir/src/base/abci/abcEco.c.o CMakeFiles/libabc.dir/src/base/abci/abcExact.c.o CMakeFiles/libabc.dir/src/base/abci/abcExtract.c.o CMakeFiles/libabc.dir/src/base/abci/abcFraig.c.o CMakeFiles/libabc.dir/src/base/abci/abcFx.c.o CMakeFiles/libabc.dir/src/base/abci/abcFxu.c.o CMakeFiles/libabc.dir/src/base/abci/abcGen.c.o CMakeFiles/libabc.dir/src/base/abci/abcHaig.c.o CMakeFiles/libabc.dir/src/base/abci/abcIf.c.o CMakeFiles/libabc.dir/src/base/abci/abcIfif.c.o CMakeFiles/libabc.dir/src/base/abci/abcIfMux.c.o CMakeFiles/libabc.dir/src/base/abci/abcIvy.c.o CMakeFiles/libabc.dir/src/base/abci/abcLog.c.o CMakeFiles/libabc.dir/src/base/abci/abcLut.c.o CMakeFiles/libabc.dir/src/base/abci/abcLutmin.c.o CMakeFiles/libabc.dir/src/base/abci/abcMap.c.o CMakeFiles/libabc.dir/src/base/abci/abcMerge.c.o CMakeFiles/libabc.dir/src/base/abci/abcMfs.c.o CMakeFiles/libabc.dir/src/base/abci/abcMini.c.o CMakeFiles/libabc.dir/src/base/abci/abcMiter.c.o CMakeFiles/libabc.dir/src/base/abci/abcMulti.c.o CMakeFiles/libabc.dir/src/base/abci/abcNtbdd.c.o CMakeFiles/libabc.dir/src/base/abci/abcNpn.c.o CMakeFiles/libabc.dir/src/base/abci/abcNpnSave.c.o CMakeFiles/libabc.dir/src/base/abci/abcOdc.c.o CMakeFiles/libabc.dir/src/base/abci/abcOrder.c.o CMakeFiles/libabc.dir/src/base/abci/abcPart.c.o CMakeFiles/libabc.dir/src/base/abci/abcPrint.c.o CMakeFiles/libabc.dir/src/base/abci/abcProve.c.o CMakeFiles/libabc.dir/src/base/abci/abcQbf.c.o CMakeFiles/libabc.dir/src/base/abci/abcQuant.c.o CMakeFiles/libabc.dir/src/base/abci/abcRec3.c.o CMakeFiles/libabc.dir/src/base/abci/abcReconv.c.o CMakeFiles/libabc.dir/src/base/abci/abcReach.c.o CMakeFiles/libabc.dir/src/base/abci/abcRefactor.c.o CMakeFiles/libabc.dir/src/base/abci/abcRenode.c.o CMakeFiles/libabc.dir/src/base/abci/abcReorder.c.o CMakeFiles/libabc.dir/src/base/abci/abcRestruct.c.o CMakeFiles/libabc.dir/src/base/abci/abcResub.c.o CMakeFiles/libabc.dir/src/base/abci/abcRewrite.c.o CMakeFiles/libabc.dir/src/base/abci/abcRpo.c.o CMakeFiles/libabc.dir/src/base/abci/abcRr.c.o CMakeFiles/libabc.dir/src/base/abci/abcRunGen.c.o CMakeFiles/libabc.dir/src/base/abci/abcSat.c.o CMakeFiles/libabc.dir/src/base/abci/abcSaucy.c.o CMakeFiles/libabc.dir/src/base/abci/abcScorr.c.o CMakeFiles/libabc.dir/src/base/abci/abcSense.c.o CMakeFiles/libabc.dir/src/base/abci/abcSpeedup.c.o CMakeFiles/libabc.dir/src/base/abci/abcStrash.c.o CMakeFiles/libabc.dir/src/base/abci/abcSw [... cut ...] d/llb/llb4Nonlin.c.o CMakeFiles/libabc.dir/src/bdd/llb/llb4Sweep.c.o
#8 592.4 /opt/rh/devtoolset-8/root/usr/bin/ranlib libabc.a
#8 592.6 gmake[2]: Leaving directory '/OpenROAD/build'
#8 592.6 [ 79%] Built target libabc
#8 592.6 gmake[1]: Leaving directory '/OpenROAD/build'
#8 592.6 gmake: *** [Makefile:139: all] Error 2
------
executor failed running [/bin/sh -c ./etc/Build.sh -compiler=${compiler} -threads=${numThreads}]: exit code: 2
```

### Additional Context

My OS is WSL2 with Ubuntu 18.04

vvbandeira, 2022-12-07:
@zerzerzerz I don't quite understand the output of your Environment section, which is failing on cmake and the Relevant log output section shows the code compiling (which is after the cmake setup). Can you try to run:
```
./build_openroad.sh --clean
```
Please upload the `build_openroad.log` file, if the file is not generated this means you are not on the latest commit.

zerzerzerz, 2022-12-07:
Thanks!
Actually, I have searched other close issues and try their methods but all failed..., including `./build_openroad.sh --clean`.
I did not find `build_openroad.log` file through vscode searching `Ctrl + P`, so I upload the `tools/OpenROAD/CMakeFiles/CMakeError.log` file.
How to use the latest commit? I run `git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts` today (2022-12-07).
My OS is WSL2 with Ubuntu-18.04.
Thanks you!
`tools/OpenROAD/CMakeFiles/CMakeError.log`
```
Performing C SOURCE FILE Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing failed with the following output:
Change Dir: /home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_1f80c/fast"
/usr/bin/make -f CMakeFiles/cmTC_1f80c.dir/build.make CMakeFiles/cmTC_1f80c.dir/build
make[1]: Entering directory '/home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_1f80c.dir/src.c.o
/usr/bin/gcc   -DC_COMPILER_SUPPORTS__-Wno-c++11-narrowing   -Wno-c++11-narrowing -o CMakeFiles/cmTC_1f80c.dir/src.c.o   -c /home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp/src.c
<command-line>: warning: ISO C99 requires whitespace after the macro name
cc1: warning: unrecognized command line option '-Wno-c++11-narrowing'
Linking C executable cmTC_1f80c
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_1f80c.dir/link.txt --verbose=1
/usr/bin/gcc  -DC_COMPILER_SUPPORTS__-Wno-c++11-narrowing    -rdynamic CMakeFiles/cmTC_1f80c.dir/src.c.o  -o cmTC_1f80c 
make[1]: Leaving directory '/home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp'

Source file was:
int main(void) { return 0; }
Performing C++ SOURCE FILE Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing failed with the following output:
Change Dir: /home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_62e6f/fast"
/usr/bin/make -f CMakeFiles/cmTC_62e6f.dir/build.make CMakeFiles/cmTC_62e6f.dir/build
make[1]: Entering directory '/home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_62e6f.dir/src.cxx.o
/usr/bin/g++    -DCXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing   -Wno-c++11-narrowing -o CMakeFiles/cmTC_62e6f.dir/src.cxx.o -c /home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp/src.cxx
<command-line>: warning: ISO C++11 requires whitespace after the macro name
cc1plus: warning: unrecognized command line option '-Wno-c++11-narrowing'
Linking CXX executable cmTC_62e6f
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_62e6f.dir/link.txt --verbose=1
/usr/bin/g++   -DCXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing    -rdynamic CMakeFiles/cmTC_62e6f.dir/src.cxx.o  -o cmTC_62e6f 
make[1]: Leaving directory '/home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp'

Source file was:
int main() { return 0; }
Performing C SOURCE FILE Test C_COMPILER_SUPPORTS__-Wno-register failed with the following output:
Change Dir: /home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_1c804/fast"
/usr/bin/make -f CMakeFiles/cmTC_1c804.dir/build.make CMakeFiles/cmTC_1c804.dir/build
make[1]: Entering directory '/home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_1c804.dir/src.c.o
/usr/bin/gcc   -DC_COMPILER_SUPPORTS__-Wno-register   -Wno-register -o CMakeFiles/cmTC_1c804.dir/src.c.o   -c /home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp/src.c
cc1: warning: command line option '-Wno-register' is valid for C++/ObjC++ but not for C
<command-line>: warning: ISO C99 requires whitespace after the macro name
Linking C executable cmTC_1c804
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_1c804.dir/link.txt --verbose=1
/usr/bin/gcc  -DC_COMPILER_SUPPORTS__-Wno-register    -rdynamic CMakeFiles/cmTC_1c804.dir/src.c.o  -o cmTC_1c804 
make[1]: Leaving directory '/home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp'

Source file was:
int main(void) { return 0; }
Performing C SOURCE FILE Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal failed with the following output:
Change Dir: /home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_032f1/fast"
/usr/bin/make -f CMakeFiles/cmTC_032f1.dir/build.make CMakeFiles/cmTC_032f1.dir/build
make[1]: Entering directory '/home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_032f1.dir/src.c.o
/usr/bin/gcc   -DC_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal   -Wno-reserved-user-defined-literal -o CMakeFiles/cmTC_032f1.dir/src.c.o   -c /home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp/src.c
<command-line>: warning: ISO C99 requires whitespace after the macro name
cc1: warning: unrecognized command line option '-Wno-reserved-user-defined-literal'
Linking C executable cmTC_032f1
[... 84 lines cut ...]
make[1]: *** [CMakeFiles/cmTC_eb942.dir/src.c.o] Error 1
make[1]: Leaving directory '/home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_eb942/fast' failed
make: *** [cmTC_eb942/fast] Error 2

Source file was:
int main(void) { return 0; }
[... cut ...]
/usr/bin/g++    -DCXX_COMPILER_SUPPORTS__c++   c++ -o CMakeFiles/cmTC_b50c1.dir/src.cxx.o -c /home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp/src.cxx
g++: error: c++: No such file or directory
CMakeFiles/cmTC_b50c1.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_b50c1.dir/src.cxx.o' failed
make[1]: *** [CMakeFiles/cmTC_b50c1.dir/src.cxx.o] Error 1
make[1]: Leaving directory '/home/zerz/projects/OpenROAD-flow-scripts/tools/OpenROAD/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_b50c1/fast' failed
make: *** [cmTC_b50c1/fast] Error 2

Source file was:
int main() { return 0; }
```

zerzerzerz, 2022-12-08:
In OpenRoad, I am in the branch `* (HEAD detached at 2a9ef5fa1)`, and I try to switch to branch `master` and pull the latest commit. But it failed with 
```
Fetched in submodule path 'tools/OpenROAD', but it did not contain 2a9ef5fa175f38558e01681e98a7572316b3b17b. Direct fetching of that commit failed.
```
If I did not switch to branch `master` in OpenROAD, it failed to pull anything.

zerzerzerz, 2022-12-08:
I found you updated OpenROAD to latest and I will try it. Thank you!

zerzerzerz, 2022-12-08:
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/10181581/build_openroad.log)

zerzerzerz, 2022-12-08:
Thanks and I will try it.

zerzerzerz, 2022-12-08:
After executing `./build_openroad.sh --clean --threads 4`, I build it successfully.
So I follow the tutorials to execute
```bash
source ./setup_env.sh
yosys -help
openroad -help
cd flow
make
exit
```
But when I execute `openroad -help`, it failed with `openroad: error while loading shared libraries: libortools.so.9: cannot open shared object file: No such file or directory`
Here is my 
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/10181932/build_openroad.log)
Thank you!

rovinski, 2022-12-08:
Are you running those commands inside the docker container? Does running `export LD_LIBRARY_PATH=/opt/or-tools/lib` help?

zerzerzerz, 2022-12-08:
I first run `docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro openroad/flow-scripts` to enter docker container.
Then I run
```bash
source ./setup_env.sh
export LD_LIBRARY_PATH=/opt/or-tools/lib
openroad -help
```
But it still failed with
```bash
openroad: error while loading shared libraries: libortools.so.9: cannot open shared object file: No such file or directory
```

vijayank88, 2022-12-08:
@zerzerzerz 
After clone goto `OpenROAD-flow-scripts/tools/OpenROAD` and run the following:
```
./etc/DependencyInstaller.sh -run
./etc/DependencyInstaller.sh -dev
```
Then
```
cd OpenROAD-flow-scripts
./build_openroad.sh
```
It will resolve libortools.so.9 issue.

zerzerzerz, 2022-12-08:
Thanks for your sincere help! I am sorry that because of my network problem, the speed accessing to or_tools from github is very slow so I cannot know whether it works for me, but I will try these steps on other machines.
I will close this issue.
Thanks again for your help!

closed 2022-12-08 by zerzerzerz

vvbandeira, 2022-12-08:
> @zerzerzerz After clone goto `OpenROAD-flow-scripts/tools/OpenROAD` and run the following:
> 
> ```
> ./etc/DependencyInstaller.sh -run
> ./etc/DependencyInstaller.sh -dev
> ```
> 
> Then
> 
> ```
> cd OpenROAD-flow-scripts
> ./build_openroad.sh
> ```
> 
> It will resolve libortools.so.9 issue.

@vijayank88 that should not be required when using docker, have you seen this on docker image before?