The-OpenROAD-Project/OpenROAD-flow-scripts #684: Performance limitation about OpenROAD tool

JiaDYuan, 2022-12-12:
### Description

Hi,
         We are using the BOOM cores  to make sure that the micro-architecture is complex enough for our research , which can produce more critical paths of interconnection. 

        I have a very concerned question about the tool OpenROAD. Harrison said that the OpenROAD now is unable to carry out the route of BOOM designs with memory. So I want to ask what is the upper performance limitation of the OpenROAD. How many gates of design can OpenROAD carry to place&route? 1 or 2 million gates? What is the largest design can run (synthesis and P&R) in the OpenROAD? Do you know the example around the approximate largest limitation?  And OpenROAD cannot do the any P&R of SRAM in any design, or just happen to BOOM cores? Most importantly, is it possible to finish the P&R of BOOM cores through OpenROAD?

maliberty, 2022-12-12:
srams are usually hard macros so I'm not sure what you are asking about.  Are you trying to create rams out of registers?

JiaDYuan, 2022-12-12:
Yes, at first, we tried to synthesize the small BOOM core with RAM, but  it is always stucking in SHARE pass stage in Yosys.

closed 2022-12-12 by maliberty

JiaDYuan, 2022-12-13:
Yes, i know there will be terrible result. What I concern is if OpenROAD can carry the synthesis and P&R process of BOOM cores. I want to know the upper performance limitation of OpenROAD. Because it is stuck in the synthesis stage without any error infos for a long time .....

maliberty, 2022-12-13:
There is no absolute limit on size - it depends on how much memory you have and how much runtime you are willing to wait for.

maliberty, 2022-12-13:
You'll have to provide a test case to show the issue.

rovinski, 2022-12-13:
It's likely not even "stuck" it is simply running it's algorithm on an unnecessarily large design. If you wait long enough it can likely finish. But how long do you want to wait? Hours? Days? Weeks?

If this is a bug with Yosys, you should probably be filing an issue on that repo. But they will likely tell you the same thing - you should be using a memory block not flip-flops.