The-OpenROAD-Project/OpenROAD-flow-scripts #689: Export STA information

zerzerzerz, 2022-12-15:
### Description

Thanks for your outstanding work! I want to know whether it exists a method to export the STA information like `.csv` format? Thanks!

maliberty, 2022-12-15:
What do you mean by "STA information" exactly?  There is no .csv export currently but a lot of information is accessible through tcl.

zerzerzerz, 2022-12-15:
After running `make gui_final`, I want to export the `Timing Report`, especially the `Data Path Details`. Thanks.

zerzerzerz, 2022-12-15:
After opening the GUI, I cannot type anything in the tcl...
I can only use my mouse to click, but I cannot enter any tcl commands...
I am using docker in ubuntu-18.04

maliberty, 2022-12-15:
I'm not aware of a way to dump out the timing paths in anything other than the ascii format report_checks uses.

@gadfort any thoughts on a dump from GUI?

@tspyrou is there any way to get a path as a tcl list like some tools provide?

maliberty, 2022-12-15:
The GUI issue is a separate problem.  I've never seen that and have to assume it is something local to your X environment.

maliberty, 2022-12-18:
You don't need the gui - just run report_checks

gadfort, 2022-12-20:
> I'm not aware of a way to dump out the timing paths in anything other than the ascii format report_checks uses.
> 
> @gadfort any thoughts on a dump from GUI?
> 
> @tspyrou is there any way to get a path as a tcl list like some tools provide?

I would think running report_checks and copying the output from the logger would be the best solution.

zerzerzerz, 2022-12-24:
Can I use `report_checks` to export all path information? Since it can only dump one path.

maliberty, 2022-12-24:
You can use -group_count <int>.  Please see the docs https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/doc/OpenSTA.pdf for more info

closed 2023-01-06 by zerzerzerz