The-OpenROAD-Project/OpenROAD-flow-scripts #732: ASAP7 has a single celler fill causing millions of instances to be created

oharboe, 2022-12-30:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

ASAP7 only has a single filler cell defined. This causes millions of instances to be created, which again causes problems for the OpenROAD GUI.

### Expected Behavior

Multiple filler cell sizes, so that so many instances doesn't have to be created.

### Environment

```shell
Git commit: d297c5b18d08684b550e74a3115d4c366bd92e01
kernel: Linux 5.19.0-26-generic
os: Ubuntu 22.10 (Kinetic Kudu)
cmake version 3.24.2
gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0
Ubuntu clang version 15.0.2-1
```

### To Reproduce

Run asap7 design vs sky130hd, look at number of filler cells created. Many more filler cells in asap7.

### Relevant log output

```shell
From 4_2_cts_fillcell.log:

[INFO DPL-0001] Placed 9362608 filler instances.
```

vijayank88, 2022-12-30:
There are two filler cells available in ASAP7.
```
FILLER_ASAP7_75t_R
FILLERxp5_ASAP7_75t_R
```
Tapcell list is:
```
TAPCELL_ASAP7_75t_R
TAPCELL_WITH_FILLER_ASAP7_75t_R
```
By replacing `FILLER_ASAP7_75t_R` the logs as follows:
```
[INFO DPL-0001] Placed 3576 filler instances.
```

oharboe, 2022-12-30:
@vijayank88 A silly question, how do I enable this in my design?

vijayank88, 2022-12-30:
Option 1: Add `export FILL_CELLS              = "FILLERxp5_ASAP7_75t_R"` in `designs/name/config.mk` and run the flow

Option 2: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/2b93d1f5f60a7e1fd2b55823379a1a485e865cec/flow/platforms/asap7/config.mk#L105 modify locally and run the flow.

maliberty, 2022-12-30:
It would be better to add them at the platform level rather than for a specific design.  I was just holding off until we have a better set of cells.

oharboe, 2022-12-30:
> It would be better to add them at the platform level rather than for a specific design. I was just holding off until we have a better set of cells.

Makes sense

maliberty, 2023-01-09:
I looked at the design you linked to from this test.  It is 2mm on a side and only a few hundred gates.  It is 99.99+% empty space.  Are you trying to test a large number of objects in the GUI?  It isn't representative otherwise.

oharboe, 2023-01-10:
> I looked at the design you linked to from this test. It is 2mm on a side and only a few hundred gates. It is 99.99+% empty space. Are you trying to test a large number of objects in the GUI? It isn't representative otherwise.

Yes: I eviscerated the design I was working on and this was all that was needed to bring the GUI to a crawl.

oharboe, 2023-01-10:
Build times have improved enormously. From 118 minutes to 50 minutes for medium sized design, as an example. For bigger designs should show a bigger improvements.

I haven't tested the GUI yet.

oharboe, 2023-01-12:
Gui works to. 

Thanks!

closed 2023-01-12 by oharboe