The-OpenROAD-Project/OpenROAD-flow-scripts #751: [INFO FLW-0008] Clock clock period -950000018091105591419833328319696937680896.000

oharboe, 2023-01-03:
### Subject

[Stage]: Global Router.

### Describe the bug

Something goes wrong with clock period calculation. 

```
[WARNING ANT-0011] -report_violating_nets is deprecated.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO ANT-0002] Found 0 net violations.
[INFO ANT-0001] Found 0 pin violations.
[INFO FLW-0007] clock clock period 250.000000
[INFO FLW-0008] Clock clock period -950000018091105591419833328319696937680896.000
[INFO FLW-0009] Clock clock slack 1000000019043269116898905829602460421849088.000
[ERROR STA-0414] period '-9.500000180911056e+41' is not a positive float.
Error: write_ref_sdc.tcl, 34 STA-0414
Elapsed time: 0:01.39[h:]min:sec. CPU time: user 1.30 sys 0.07 (99%). Peak memory: 243664KB.
make: *** [results/asap7/mock-array-big_Element/base/5_1_grt.odb] Error 1
```

### Expected Behavior

The slack should be in some reasonable range.

### Environment

```shell
Git commit: bb4801b7457d04f2ddf27c202803dfbe5c8f324d
kernel: Linux 5.15.0-56-generic
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.24.2
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Ubuntu clang version 14.0.0-1ubuntu1
```

### To Reproduce

Using https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/750, run:

```shell
make DESIGN_CONFIG=designs/asap7/mock-array-big/Element/config.mk
[deleted]
[WARNING ANT-0011] -report_violating_nets is deprecated.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO ANT-0002] Found 0 net violations.
[INFO ANT-0001] Found 0 pin violations.
[INFO FLW-0007] clock clock period 250.000000
[INFO FLW-0008] Clock clock period -950000018091105591419833328319696937680896.000
[INFO FLW-0009] Clock clock slack 1000000019043269116898905829602460421849088.000
[ERROR STA-0414] period '-9.500000180911056e+41' is not a positive float.
Error: write_ref_sdc.tcl, 34 STA-0414
Elapsed time: 0:01.39[h:]min:sec. CPU time: user 1.30 sys 0.07 (99%). Peak memory: 243664KB.
make: *** [results/asap7/mock-array-big_Element/base/5_1_grt.odb] Error 1

```

rovinski, 2023-01-05:
Relevant https://github.com/The-OpenROAD-Project/OpenROAD/issues/2268

maliberty, 2023-01-09:
Your block has no constrained paths.  I'll fix the code to skip in this case but your test case needs fixing to be representative.

oharboe, 2023-01-09:
> Your block has no constrained paths. I'll fix the code to skip in this case but your test case needs fixing to be representative.

Silly question: what does that mean?

I have a clock, a register and a constraint file and the name of the clock is "clock".

I would have thought that I  then do have constrained paths.

maliberty, 2023-01-09:
You have no constraints on your IOs and no reg-to-reg paths.

oharboe, 2023-01-09:
> You have no constraints on your IOs and no reg-to-reg paths.

I see. I need help setting up those constraints.sdc files obviously. Will check with colleague.

PR #763 (2023-01-11): Skip writing the updated_clk.sdc if there is no constrained path
Fixes #751
commit 868baf69 (2023-01-11): Skip writing the updated_clk.sdc if there is no constrained path

Fixes #751
closed 2023-01-11 by maliberty