The-OpenROAD-Project/OpenROAD-flow-scripts #772: Command could not found openroad

JeffreyHJF, 2023-01-17:
### Subject

[Build] for compilation-related issues.

### Describe the bug

"openroad" command cannot be used after installation of the project.
I try to find command file to link to my environment but it does not work.,
I cannot find any bin or sbin/openroad file in my OpenROAD. Only a file called openroad in the direction of ./src/openroad.

### Expected Behavior

I need to run the openroad command to start a design flow.

### Environment

```shell
Ubuntu 22.04
```

### To Reproduce

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git

cd OpenROAD
sudo ./etc/DependencyInstaller.sh -run
sudo ./etc/DependencyInstaller.sh -dev
sudo apt-get install libspdlog-dev
sudo apt install tcl-dev
./etc/Build.sh

openroad -gui # the final row

error then happening

vijayank88, 2023-01-17:
@JeffreyHJF 
Can you check the following path for `openroad` executable?
`OpenROAD/build/src`

vijayank88, 2023-01-17:
Other way add following format in `.bashrc` file  and execute from anywhere
`export OPENROAD_HOME="<path-to>/OpenROAD/build/src"`
`source ~/.bashrc`
`$OPENROAD_HOME/openroad -gui`

JeffreyHJF, 2023-01-17:
> @JeffreyHJF Can you check the following path for `openroad` executable? `OpenROAD/build/src`

yeah I have try that, it does not work. Thanks for your help I would take the advice to try again

JeffreyHJF, 2023-01-17:
at my home page, I used **vi ~ /.bashrc** to go into the file you mentioned and add the content you said, but finally, when I used the command of openroad, it is still not found.
I don't know if I missed any point in the operation

vijayank88, 2023-01-17:
Can you try below command once again in new directory and let me know
```
cd
mkdir OPENROAD
cd OPENROAD
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
cd OpenROAD
./etc/DependencyInstaller.sh -local
./etc/Build.sh
```
Then try from `src` directory to lanuch `openroad -gui`

JeffreyHJF, 2023-01-17:
eeuser@eeuser-HP-EliteDesk-800-G3-TWR:~/OPENROAD/OpenROAD/src$ openroad -gui
openroad: command not found

I have repeated the steps you mentioned, but cannot open the GUI

vijayank88, 2023-01-17:
Try this way  `./openroad -gui`