The-OpenROAD-Project/OpenROAD-flow-scripts #773: Unable to complete docker installation with MacOS Air M1

vijayank88, 2023-01-17:
### Subject

[Build] for compilation-related issues.

### Describe the bug

While trying install ORFS with docker failed to complete. 

### Expected Behavior

Complete ORFS installation in MacOS docker mode as well

### Environment

```shell
MacOS Air M1
chip Apple M1
Version 12.5
```

### To Reproduce

Followed below steps in MacOS
```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh
```

### Relevant log output

```shell
tee:build_openroad.log: permission denied

#8 1643.1 [100%] Building yosys
#8 1661.5 mkdir -p /install/bin
#8 1661.6 cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc /install/bin
#8 1661.8 strip -S /install/bin/yosys
#8 1662.0 strip /install/bin/yosys-abc
#8 1662.0 strip:/install/bin/yosys-abc: File format not recognized
#8 1662.0 make: *** [Makefile:859: install] Error 1
#8 ERROR: executor failed running [/bin/sh -c make PREFIX=/install CONFIG=gcc ABCREV=bafd2a7 ABCURL=https://github.com/berkeley-abc/abc install -j$(nproc)]: exit code: 2
------
 > [4/4] RUN make PREFIX=/install CONFIG=gcc ABCREV=bafd2a7 ABCURL=https://github.com/berkeley-abc/abc install -j$(nproc):
------
executor failed running [/bin/sh -c make PREFIX=/install CONFIG=gcc ABCREV=bafd2a7 ABCURL=https://github.com/berkeley-abc/abc install -j$(nproc)]: exit code: 2
```

vvbandeira, 2023-01-17:
Same a https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/767#issuecomment-1379382591. Make sure the workspace is clean and upload build_openroad.log.

> `tee:build_openroad.log: permission denied`

The above would also suggest the user tried to compile once with sudo and once without, which is a problem.

tsaoalbert, 2023-01-18:
I've made fresh clone in new directory and update you on this again

tsaoalbert, 2023-01-18:
The build still failed as shown in the log below.

[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/10445657/build_openroad.log)

vijayank88, 2023-01-18:
We see progress in installation.  Now it fails due to `nproc` not found in MacOS.
```
Use 'docker scan' to run Snyk tests against images to find vulnerabilities and learn how to fix them
[INFO FLW-0022] Building docker image for OpenROAD app.
./tools/OpenROAD/etc/DockerHelper.sh: line 275: nproc: command not found
```
I found following equivalent command.

`sysctl -n hw.physicalcpu` is the equivalent to `nproc`  on macOS.

@vvbandeira Can we do `alias nproc="sysctl -n hw.logicalcpu"` and continue?

vvbandeira, 2023-01-18:
> @vvbandeira Can we do `alias nproc="sysctl -n hw.logicalcpu"` and continue?
> or use `./build_openroad.sh --threads 4`

This should work for now. @habibayassin could you make a PR to improve support of the `build_openroad.sh` with macOS?

habibayassin, 2023-01-20:
> The build still failed as shown in the log below.
> 
> [build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/10445657/build_openroad.log)

@vvbandeira I opened [this PR](https://github.com/The-OpenROAD-Project/OpenROAD/pull/2798) in OpenROAD as the above log file is generating an error from  `./tools/OpenROAD/etc/DockerHelper.sh`

vvbandeira, 2023-01-23:
Thanks @habibayassin. I just merged the PR. @tsaoalbert, could you get the latest code a try?

vijayank88, 2023-01-24:
@vvbandeira Can you update `OpenROAD` submodule as well to reflect the changes?

habibayassin, 2023-01-24:
> @vvbandeira Can you update `OpenROAD` submodule as well to reflect the changes?

@vvbandeira  @vijayank88 I have an open [PR](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/697) that has the updated OR submodule ready to be merged

vijayank88, 2023-02-01:
@vvbandeira 
Can you please review #697  and merge it.?
User waiting for a long. Tried with latest master for docker mode, still facing issue with `nproc`
```
#10 exporting to image
#10 sha256:e8c613e07b0b7ff33893b694f7759a10d42e180f2b4dc349fb57dc6b71dcab00
#10 exporting layers
#10 exporting layers 2.7s done
#10 writing image sha256:908d45cee2d6c09f446ba7fde18eca8479f960547977819d2c4a59d925980218 done
#10 naming to [docker.io/openroad/lsoracle](http://docker.io/openroad/lsoracle) done
#10 DONE 2.7s

Use 'docker scan' to run Snyk tests against images to find vulnerabilities and learn how to fix them
[INFO FLW-0022] Building docker image for OpenROAD app.
./tools/OpenROAD/etc/DockerHelper.sh: line 275: nproc: command not found
```

vijayank88, 2023-02-15:
@vvbandeira 
It's got stuck during docker pull at the end:
```
#6 FROM docker.io/openroad/yosys:latest
#6 sha256:6511b9f1389b053b4083f865107faf48eb4c4b0d901e54f8b1add3a37db9e34c
#6 resolve docker.io/openroad/yosys:latest
#6 ...

#8 FROM docker.io/openroad/lsoracle:latest
#8 sha256:312c4b5a4fd35ef8a964618cd2cac3ad676415074c0a25a11d350212c28701ba
#8 resolve docker.io/openroad/lsoracle:latest 1.0s done
#8 ERROR: pull access denied, repository does not exist or may require authorization: server message: insufficient_scope: authorization failed

#12 FROM docker.io/openroad/centos7-builder-gcc:latest
#12 sha256:227382524c70140297a556f99bdfa825fa921d093e7486866946b0759202ef90
#12 resolve docker.io/openroad/centos7-builder-gcc:latest 1.1s done
#12 ERROR: pull access denied, repository does not exist or may require authorization: server message: insufficient_scope: authorization failed

#6 FROM docker.io/openroad/yosys:latest
#6 sha256:6511b9f1389b053b4083f865107faf48eb4c4b0d901e54f8b1add3a37db9e34c
#6 resolve docker.io/openroad/yosys:latest 1.2s done
#6 CANCELED
------
 > FROM docker.io/openroad/lsoracle:latest:
------
------
 > FROM docker.io/openroad/centos7-builder-gcc:latest:
------
failed to load cache key: pull access denied, repository does not exist or may require authorization: server message: insufficient_scope: authorization failed
```
How to fix the above error?

vvbandeira, 2023-02-15:
please upload the full `build_openroad.log`

vijayank88, 2023-02-15:
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/10743468/build_openroad.log)

vvbandeira, 2023-02-21:
could you try again with the latest code? I have updated the image upstream.