The-OpenROAD-Project/OpenROAD-flow-scripts #780: Issue when testing Design flow with Ibex

tristonw109, 2023-01-20:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

The flow runs all the way until it runs into KLayout. Here is the exact output:

### Expected Behavior

I have no idea what to expect after this because this is the first time using OpenROAD flow. 

### Environment

```shell
Please pull the latest changes and try again, if problem persists file a GitHub issue with the re-producible test case.

//I will note that after receiving the message, I pulled and received this message:

Already up to date.
```

### To Reproduce

Simply run the following command:

`
make DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk
`

Will note that I have KLayout v0.27.9. I tried the suggested v0.27.1 and the error persists.

### Relevant log output

```shell
[INFO][FLOW] Using platform directory ./platforms/sky130hd
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' stdbuf -o L /usr/bin/klayout -zz -rd design_name=ibex_core \
        -rd in_def=results/sky130hd/ibex/base/6_final.def \
        -rd in_files="./platforms/sky130hd/gds/sky130_fd_sc_hd.gds  " \
        -rd config_file=./platforms/sky130hd/fill.json \
        -rd seal_file="" \
        -rd out_file=results/sky130hd/ibex/base/6_1_merged.gds \
        -rd tech_file=./objects/sky130hd/ibex/base/klayout.lyt \
        -rd layer_map= \
        -r ./util/def2stream.py) 2>&1 | tee ./logs/sky130hd/ibex/base/6_1_merge.log
ERROR: Signal number: 11
Address: 0x80
Program Version: KLayout 0.27.9 (2022-04-24 r66c0eacca)

Backtrace:
/usr/lib/klayout/libklayout_lay.so.0 +0x2c9e87 lay::enable_signal_handler_gui(bool) [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x42520 __restore_rt [libc_sigaction.c:?]
/usr/lib/klayout/libklayout_tl.so.0 +0x99288 tl::Variant::operator<(tl::Variant const&) const [??:?]
/usr/lib/klayout/libklayout_db.so.0 +0x798ea7 db::PropertiesRepository::prop_name_id(tl::Variant const&) [??:?]
/usr/bin/db_plugins/liblefdef.so +0xa5bca db::LEFDEFImporter::read(tl::InputStream&, db::Layout&, db::LEFDEFReaderState&) [??:?]
/usr/bin/db_plugins/liblefdef.so +0x63eef db::LEFDEFReader::read_lefdef(db::Layout&, db::LoadLayoutOptions const&, bool) [??:?]
/usr/bin/db_plugins/liblefdef.so +0x65a03 db::LEFDEFReader::read(db::Layout&, db::LoadLayoutOptions const&) [??:?]
/usr/lib/klayout/libklayout_db.so.0 +0x79ce92 db::Reader::read(db::Layout&, db::LoadLayoutOptions const&) [??:?]
/usr/lib/klayout/libklayout_db.so.0 +0xe89286 gsi::VectorAdaptorImpl<std::vector<db::polygon<double>, std::allocator<db::polygon<double> > > >::push(gsi::SerialArgs&, tl::Heap&) [??:?]
/usr/lib/klayout/libklayout_db.so.0 +0xe93c7e void gsi::MethodBase::add_arg<db::LayerProperties const&>(gsi::ArgSpecBase const&) [??:?]
/usr/lib/klayout/libklayout_pya.so.0 +0x5a4cc pya::PythonModule::init(char const*, _object*) [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0xf021e PyComplex_AsCComplex [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x79eee _PyEval_EvalFrameDefault [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c681f _PyEval_SetAsyncGenFinalizer [??:?]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0 +0x1c194e PyEval_EvalCode [??:?]
/usr/lib/klayout/libklayout_pya.so.0 +0x3a6c7 pya::PythonInterpreter::eval_string(char const*, char const*, int, int) [??:?]
/usr/lib/klayout/libklayout_lym.so.0 +0x4f7ef lym::Macro::run() const [??:?]
/usr/lib/klayout/libklayout_lay.so.0 +0x146f58 lay::ApplicationBase::run() [??:?]
/usr/bin/klayout +0x4a9d ?? [??:0]
/usr/lib/klayout/libklayout_rba.so.0 +0x23834 rba::RubyInterpreter::set_debugger_scope(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) [??:?]
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x22fc35 rb_dtrace_setup [??:?]
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x231fc6 rb_vm_localjump_error [??:?]
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x234c32 rb_check_funcall [??:?]
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23a30d rb_vm_exec [??:?]
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0xa5318 rb_check_copyable [??:?]
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0xa8e1a ruby_run_node [??:?]
/usr/lib/klayout/libklayout_rba.so.0 +0x24b52 rba::RubyInterpreter::initialize(int&, char**, int (*)(int&, char**)) [??:?]
/usr/bin/klayout +0x46f2 ?? [??:0]
/usr/bin/klayout +0x3ed7 ?? [??:0]
/lib/x86_64-linux-gnu/libc.so.6 +0x29d90
/lib/x86_64-linux-gnu/libc.so.6 +0x29e40 __libc_start_main
/usr/bin/klayout +0x3f95
Command exited with non-zero status 11
Elapsed time: 0:00.80[h:]min:sec. CPU time: user 0.74 sys 0.05 (99%). Peak memory: 369368KB.
make: *** [Makefile:685: results/sky130hd/ibex/base/6_1_merged.gds] Error 11
```

maliberty, 2023-01-20:
Are you able to run klayout outside the flow?  We test this design nightly with klayout and don't see this crash so it must be something local to your setup.

tristonw109, 2023-01-20:
> Are you able to run klayout outside the flow? We test this design nightly with klayout and don't see this crash so it must be something local to your setup.

I can launch klayout outside the flow, however, I get this warning:
```
Info: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
```

I am currently testing a clean set-up in a VM with Ubuntu 20.04.5. I can use klayout without any warning, which I believe to be a good sign. But I am faced with this syntax error: 

```
awk: line 1: syntax error at or near ,
make: *** [Makefile:640: finish] Error 2
```
The Line in question:
```
@grep Elapsed $(abspath $(LOG_DIR))/*.log | awk 'match($$0, /[a-zA-Z_0-9/]+\/([a-zA-Z_0-9]+)\.log:Elapsed time: ([09]+:)?([0-9]+):([0-9]+)/ , arr) { printf "%-25s %10d\n", arr[1], arr[2]*3600 + arr[3] * 60 + arr[4] }'
```

maliberty, 2023-01-21:
What version of awk do you have?  It seems rather odd as awk doesn't change often.

tristonw109, 2023-01-21:
Fixed the error. Needed to update to gawk. Going to chalk up the original issue being related to Ubuntu 22.04. OpenROAD runs well on Ubuntu 20.04.5. I appreciate your guidance!

closed 2023-01-21 by tristonw109