The-OpenROAD-Project/OpenROAD-flow-scripts #790: ASAP7 GDS appears to be partially shrunk 4x

antonblanchard, 2023-01-27:
### Subject

[Stage]: GDS generation.

### Describe the bug

I was looking at the ASAP7 GDS produced by ORFS and it looks a bit like the wires were shrunk 4x, but the standard cell contents were not.

M1:
[image]

M2:
[image]

If I load the def into klayout, the M2 pitch looks good:

[image]

If I look at the M2 pitch in the GDS, it's 1/4 size:

[image]

I'm not sure what is causing this, I immediately thought of the ASAP7 4x hack, but we are using the 1x def.

An example of the issue: https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Public/job/public_tests_all/job/master/594/artifact/flow/final_report_gcd_asap7_base_2023-01-25_15-42.tar.gz

### Expected Behavior

Correct sized GDS and no overlapping issues in M1/M2

### Environment

```shell
Git commit: 40bbbab3b1a446342a6341021a27a2a8b07d8d8f
os: Fedora Linux 37 (Workstation Edition)
cmake version 3.25.1
gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4)
clang version 15.0.7 (Fedora 15.0.7-1.fc37)
```

### To Reproduce

Run any ASAP7 test in the CI

maliberty, 2023-01-27:
That is really odd and I don't see an obvious explanation.   @rovinski any clues?

antonblanchard, 2023-01-27:
The ASAP7 standard cell GDS libraries have `UNITS = 0.00025, 2.5e-10`, vs say the sky130hd ones which have `UNITS = 0.001, 1e-09`. That may have something to do with it.

maliberty, 2023-01-28:
Thanks, that looks to be the key insight.

PR #793 (2023-01-28): Fix dbu units in asap7 to get correctly scaled gds
Fixes #790
commit 7d2ec9ac (2023-01-28): Fix dbu units in asap7 to get correctly scaled gds

Fixes #790
closed 2023-01-28 by maliberty