The-OpenROAD-Project/OpenROAD-flow-scripts #815: Make PDN config for mock-array-big design

yupferris, 2023-02-09:
### Description

Looking at the `mock-array-big` design, we see that the power stripes don't line up between the inner macros and the top-level macro, as seen below for M5:

[image]

### Suggested Solution

Other designs, such as `chameleon_hier`, have specific PDN config files for macros in the hierarchy. It would be useful to see a similar set of configs for the `mock-array-big` design.

### Additional Context

See additional relevant discussion here: https://github.com/The-OpenROAD-Project/OpenROAD/discussions/2868

maliberty, 2023-02-10:
Based on the discussion is this still relevant?

yupferris, 2023-02-23:
> Based on the discussion is this still relevant?

I don't think so.

closed 2023-02-23 by yupferris