The-OpenROAD-Project/OpenROAD-flow-scripts #855: I have an issue while installing open road with open road flow scripts

paramsaini, 2023-03-08:
### Subject

[Documentation] for documentation errors.

### Describe the bug

[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/10925811/build_openroad.log)

### Expected Behavior

I expected installation of openroad software

### Environment

```shell
(base) paramsaini@MacBook-Pro openroad % ./etc/Env.sh
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Darwin 22.3.0
os: 		macOS 		13.2.1
cmake version 3.25.3
CMake Warning at src/CMakeLists.txt:225 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

CMake Error at src/mpl2/CMakeLists.txt:42 (find_package):
  By not providing "Findortools.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ortools", but
  CMake did not find one.

  Could not find a package configuration file provided by "ortools" with any
  of the following names:

    ortoolsConfig.cmake
    ortools-config.cmake

  Add the installation prefix of "ortools" to CMAKE_PREFIX_PATH or set
  "ortools_DIR" to a directory containing one of the above files.  If
  "ortools" provides a separate development package or SDK, be sure it has
  been installed.

-- The CXX compiler identification is AppleClang 14.0.0.14000029
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-6895-g5c85b36f8
-- System name: Darwin
-- Compiler: AppleClang 14.0.0.14000029
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is AppleClang 14.0.0.14000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Success
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive - Success
-- Performing Test C_COMPILER_SUPPORTS__-x
-- Performing Test C_COMPILER_SUPPORTS__-x - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-x
-- Performing Test CXX_COMPILER_SUPPORTS__-x - Failed
-- Performing Test C_COMPILER_SUPPORTS__c++
-- Performing Test C_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__c++
-- Performing Test CXX_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- TCL library: /usr/local/lib/libtcl8.6.dylib
-- TCL header: TCL_HEADER-NOTFOUND
-- Found SWIG: /usr/local/bin/swig (found suitable version "4.1.1", minimum required is "3.0")  
-- Using SWIG >= 4.1.1 -flatstaticmethod flag for python
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0")  
-- boost: 1.81.0
-- Found Python3: /Library/Frameworks/Python.framework/Versions/3.10/include/python3.10 (found version "3.10.0") found components: Development Development.Module Development.Embed 
-- Found ZLIB: /Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/lib/libz.tbd (found version "1.2.11") 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- spdlog: 1.9.2
-- Found BISON: /usr/bin/bison (found version "2.3") 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- STA version: 2.3.3
-- STA git sha: 2a2b5cd59f5d1b596f20cbd39609c52b33299e06
-- System name: Darwin
-- Compiler: AppleClang 14.0.0.14000029
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- TCL library: /usr/local/lib/libtcl8.6.dylib
-- TCL header: TCL_HEADER-NOTFOUND
-- SSTA: 0
-- STA executable: /Users/paramsaini/openroad-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- GPU is not enabled
-- Configuring incomplete, errors occurred!
(base) paramsaini@MacBook-Pro openroad %
```

### To Reproduce

please help.........

paramsaini, 2023-03-08:
I updated the latest openroad version now but still it shows the same above log report.

maliberty, 2023-03-09:
Have you installed or-tools?

paramsaini, 2023-03-09:
still not

maliberty, 2023-03-09:
> still not

what does that mean?

paramsaini, 2023-03-09:
ERROR: Could not find a version that satisfies the requirement or-tools (from versions: none)
ERROR: No matching distribution found for or-tools

paramsaini, 2023-03-09:
I installed or tools but when I am trying to pass the command ./etc/DependencyInstaller.sh

it shows this error:

what does it mean "wget" is it a package or any file?

==> Running `brew cleanup spdlog`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
Removing: /Users/paramsaini/Library/Caches/Homebrew/spdlog--1.9.2... (328.0KB)
Warning: HOMEBREW_NO_INSTALLED_DEPENDENTS_CHECK is set: not checking for outdated
dependents or dependents with broken linkage!
./tools/OpenROAD/etc/DependencyInstaller.sh: line 139: wget: command not found
(base) paramsaini@MacBook-Pro openroad-flow-scripts %

paramsaini, 2023-03-09:
@maliberty 
thanks for your help.
can you please tell me how to remove the above error.

maliberty, 2023-03-09:
@vvbandeira would you help out

paramsaini, 2023-03-09:
here the log file:

[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/10934646/build_openroad.log)

vvbandeira, 2023-03-13:
@paramsaini you are likely running out of resources. Can try to increase the amount of RAM and CPUs in the Docker settings or run:
```
./build_openroad.sh --threads 1
```

paramsaini, 2023-03-13:
@vvbandeira thanks for your reply. I actually tried both ways to increase capacity of RAM and CPU in the docker and also run command which you mentioned above but it doesn't work. in both times I got this error.

"#8 1799.3 [output clipped, log limit 2MiB reached]"

after this processing stops.

can you please tell me how to fix this error and what does this error mean.

vijayank88, 2023-03-14:
@paramsaini 
Whether it is still on same message or got completed?
For me able to install in Centos07 with docker mode.
```
#8 314.1 [output clipped, log limit 2MiB reached]
#8 DONE 441.7s

#9 exporting to image
#9 exporting layers
#9 exporting layers 14.2s done
#9 writing image sha256:e61680a26f242446796e75b9f727d4f099d426ce done
#9 naming to docker.io/openroad/flow-centos7-builder:latest done
#9 DONE 14.2s
```

paramsaini, 2023-03-18:
@vijayank88 thanks buddy I installed it again but locally now it works.

closed 2023-03-18 by vijayank88