The-OpenROAD-Project/OpenROAD-flow-scripts #860: Pin clk does not have geometries in a valid routing layer

b224hisl, 2023-03-13:
### Subject

[Stage]: Global Router.

### Describe the bug

I'm running openroad-flow-scripts for hardening riscv32i's regfile by ASPD7, but I meet an error of `Pin clk does not have geometries in a valid routing layer`

### Expected Behavior

harden the design successfully

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.0-47-generic
os: Ubuntu 20.04.5 LTS (Focal Fossa)
cmake version 3.24.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-6895-g5c85b36f8
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
[... 417 lines cut ...]
[INFO GPL-0103] Weighted 623 nets.
[NesterovSolve] Iter: 340 overflow: 0.619791 HPWL: 15049372
[NesterovSolve] Snapshot saved at iter = 344
[NesterovSolve] Iter: 350 overflow: 0.570114 HPWL: 14568971
[NesterovSolve] Iter: 360 overflow: 0.522921 HPWL: 14502048
[NesterovSolve] Iter: 370 overflow: 0.479291 HPWL: 14455442
[INFO GPL-0100] worst slack 9.74e-10
[INFO GPL-0103] Weighted 623 nets.
[NesterovSolve] Iter: 380 overflow: 0.426763 HPWL: 14675793
[NesterovSolve] Iter: 390 overflow: 0.385195 HPWL: 14618514
[NesterovSolve] Iter: 400 overflow: 0.335492 HPWL: 14697102
[NesterovSolve] Iter: 410 overflow: 0.30819 HPWL: 14740948
[INFO GPL-0100] worst slack 9.73e-10
[INFO GPL-0103] Weighted 623 nets.
[NesterovSolve] Iter: 420 overflow: 0.274918 HPWL: 14751720
[NesterovSolve] Iter: 430 overflow: 0.247963 HPWL: 14799351
[NesterovSolve] Iter: 440 overflow: 0.218794 HPWL: 14882698
[INFO GPL-0100] worst slack 9.53e-10
[INFO GPL-0103] Weighted 623 nets.
[INFO GPL-0075] Routability numCall: 1 inflationIterCnt: 1 bloatIterCnt: 0
[ERROR GRT-0042] Pin clk does not have geometries in a valid routing layer.
Error: global_place.tcl, 49 GRT-0042
Command exited with non-zero status 1
Elapsed time: 0:20.32[h:]min:sec. CPU time: user 20.17 sys 0.14 (99%). Peak memory: 266708KB.
```

vijayank88, 2023-03-13:
@b224hisl 
Are you running automated flow and running individual stages?
Can you explain how you configured and executing the flow?
I am not seeing any pins placed around the core area.

b224hisl, 2023-03-13:
I run the automated flow, I just trying to harden the riscv32's regfile.v.

vijayank88, 2023-03-13:
missing config.mk

vijayank88, 2023-03-13:
AFAIK there is no macro hardening feature in ORFS flow.
@maliberty 
What is the use of `IS_CHIP` variable?

closed 2023-03-13 by vijayank88

b224hisl, 2023-03-13:
I remove `IS_CHIP`, but still get the same error

b224hisl, 2023-03-13:
OK I solved this problem, i have to `make clean_all` first. Thank you a lot!