The-OpenROAD-Project/OpenROAD-flow-scripts #873: Remove manual placement in design/asap7/mock-array-big and use MPL2

oharboe, 2023-03-16:
### Description

Can MPL2 lay out an array of macros?

### Suggested Solution

Remove manual placement in design/asap7/mock-array-big and switch to mpl2

maliberty, 2023-03-16:
The simplest is to try.  My guess is it won't be as perfect as doing it by yourself.

oharboe, 2023-03-16:
> The simplest is to try. My guess is it won't be as perfect as doing it by yourself.

Is mpl2 enabled by default?

Can mpl2 be extended by algorithms or heuristics to be able to handle straightforward array cases?

maliberty, 2023-03-16:
The goal is to swap over all test cases to mpl2 and retire mpl.  We aren't quite there yet.  

How would mpl2 know that is a "straightforward array case" from the netlist?

maliberty, 2024-03-19:
It is still a goal for mpl2 to handle mock-array