The-OpenROAD-Project/OpenROAD-flow-scripts #876: Unable to run make command inside docker

vijayank88, 2023-03-18:
@vvbandeira @habibayassin 
I have installed ORFS using docker mode.
I ran following commands:
```
export OS_NAME=centos7
docker run -it -u $(id -u ${USER}):$(id -g ${USER}) openroad/flow-$OS_NAME-builder
source setup_env.sh
cd flow
make
```
Failed with below message:
```
[INFO][FLOW] Using platform directory ./platforms/nangate45
mkdir: cannot create directory './objects': Permission denied
make: *** [objects/nangate45/gcd/base/lib/NangateOpenCellLibrary_typical.lib] Error 1
bash-4.2$ ls
```
How to resolve this permission issue?
Following instruction from [here](https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildWithDocker.html#verify-installation)

_Originally posted by @vijayank88 in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/862_

vijayank88, 2023-03-18:
@habibayassin 
As you're updated the documentation with docker command, can you please look into this issue?

habibayassin, 2023-03-18:
@vijayank88 yes sure I'll look into it