The-OpenROAD-Project/OpenROAD-flow-scripts #913: routing congestion too high at global placement step

b224hisl, 2023-03-23:
### Subject

[Stage]: Global Placement.

### Describe the bug

I meet the error of routing congestion even I provide `export DIE_AREA = 0 0 3000 3000`. Do you have any idea to solve this problem?

I run the flow using the command `make DESIGN_CONFIG=./designs/sky130hd/single_router/config.mk generate_abstract`.  Is there anything wrong?

### Expected Behavior

finish routing step

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.0-1028-aws
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-6895-g5c85b36f8
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
[... 128 lines cut ...]
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical            0             0          0.00%
met1       Horizontal     242000        114621          52.64%
met2       Vertical       181500        118895          34.49%
met3       Horizontal     121000         82149          32.11%
met4       Vertical        72600         35534          51.06%
met5       Horizontal      24200         12208          49.55%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[WARNING GRT-0230] Congestion iterations cannot increase overflow, reached the maximum number of times the total overflow can be increased.
[INFO GRT-0197] Via related to pin nodes: 145038
[INFO GRT-0198] Via related Steiner nodes: 5541
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 176584
[INFO GRT-0112] Final usage 3D: 731437
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load ./reports/sky130hd/single_router/b_regfile/congestion.rpt in the DRC viewer.
Error: global_route.tcl, 21 GRT-0119
Command exited with non-zero status 1
Elapsed time: 1:45.54[h:]min:sec. CPU time: user 86.81 sys 0.17 (82%). Peak memory: 301840KB.
make: *** [Makefile:593: results/sky130hd/single_router/b_regfile/5_1_grt.odb] Error 1
```

vijayank88, 2023-03-23:
@b224hisl 
The subject is wrong,  `routing congestion too high at global route step`
Share your `config.mk`
You're saying die area 3000um also causing the issue. But in test case I see only 760um only.

vijayank88, 2023-03-23:
@b224hisl 
check `reports/.../congestion.rpt` to find where overflow occurs.
Relax the placement density/ increase core area to resolve the issue.
[image]

vijayank88, 2023-07-17:
@b224hisl 
Is that resolved? Can we close this?

b224hisl, 2023-07-17:
oh sorry，i forget this issue, thank you for you help, now you can close this

closed 2023-07-17 by vijayank88