The-OpenROAD-Project/OpenROAD-flow-scripts #916: Changing macro placement causes: Maze Route cannot find path of net dmem/inter_dmem2 in worker of routeBox ( 79380 7560 ) ( 83160 11340 ).

vijayank88, 2023-03-23:
Changing macro placement causes: Maze Route cannot find path of net dmem/inter_dmem2 in worker of routeBox ( 79380 7560 ) ( 83160 11340 ).
Have tried changing top routing layer to M8
Also, `set_global_routing_layer_adjustment $env(MIN_ROUTING_LAYER)-$env(MAX_ROUTING_LAYER) 0.5`
And, `set_global_routing_layer_adjustment $env(MIN_ROUTING_LAYER)-$env(MAX_ROUTING_LAYER) 0.65`
Didn't help

To recreate the problem:

Design_nickname: riscv32i
Die_area: 0 0 120 120
Core_area: 5 5 110 110
Note: nothing changed in design/asap7/riscv32i/config.mk 

Macro placement target
Macro 1: name: dmem/dmem0, location: x0=15024 y0=10000 orientation: R0, place_status: FIRM
Macro 2: name: dmem/dmem1, location: x0=27505 y0=10000 orientation: R0, place_status: FIRM
Macro 3: name: dmem/dmem2, location: x0=82386 y0=10000 orientation: R0, place_status: FIRM
Macro 4: name: dmem/dmem3, location: x0=95867 y0=10000 orientation: R0, place_status: FIRM

_Originally posted by @Fahim36201 in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/911_

vijayank88, 2023-03-23:
@maliberty 
Here is another issue reported by user.
If automatic macro placement or Hier-RTLMP used not causing any issues during routing.
If try to do manual macro placement it throws an error:
```
[ERROR DRT-0255] Maze Route cannot find path of net dmem/we_mem\[2\] in worker of routeBox ( 79380 68040 ) ( 83160 71820 ).
```
Test case available in #911

maliberty, 2023-03-23:
The test case from the gdrive link reproduces the problem.

osamahammad21, 2023-08-15:
#911 is closed, should we close this one as well?

vijayank88, 2023-08-17:
@osamahammad21 
#911 reported that manual macro placement throw this error. So discussion converted into an issue. It is not yet resolved.

vijayank88, 2023-08-17:
If I use mpl based co-ordinates in `macro.cfg` flow completes successfully.
```
dmem.dmem0 R0 51.12 39.792
dmem.dmem1 R0 36.72 39.792
dmem.dmem2 R0 22.368 39.792
dmem.dmem3 R0 8.016 39.792
```
If I use any other values facing issue during detail routing stage.

maliberty, 2023-08-17:
@vijayank88 if you manually place macros with the pins off grid you will have problems.  We've recently added a check for that.

vijayank88, 2023-08-18:
In gui is there a way to find it, we placed macros pins ongrid? 
From gui display viewer options

maliberty, 2023-08-18:
I usually turn on the single layer with the pins and turn on the tracks.  The tracks should go through the pins' centerline.  Its not perfect but it catches obvious errors.

vijayank88, 2023-08-19:
> I usually turn on the single layer with the pins and turn on the tracks. The tracks should go through the pins' centerline. Its not perfect but it catches obvious errors.

Thanks @maliberty . After moving moving as per above instructions, it is resolved `Maze Route cannot find path of net ` error and facing DRC error post detail routing.

vijayank88, 2023-08-24:
@osamahammad21 Can you please have a look into this detail route test case. https://slack-files.com/T01699QAZBQ-F05NWHBHH1D-9c9cd46fe8 
With automatic macro placement routing completes withing 5 iteration.
With manual macro placement getting lot of DRC violations. Currently test case attached with 10 iteration. 
Even 64 iteration same number of DRC violations exist.

osamahammad21, 2023-08-25:
Could you please check if the macro pins are aligned with the routing tracks. If they are not, then there is no escape from hard routing with the current state of the router. Check in the logs for warnings with ID 418 and 419 if you are using latest updated branch.

vijayank88, 2023-08-26:
As per Matt suggestion placed macros and its pin goes through M1 tracks and M5 pin layers. Screenshot attached.
[image]

Yes warning shows in routing log as follows:
```

[WARNING DRT-0418] Term dmem.dmem0/rd_out[0] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[1] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[2] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[3] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[4] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[5] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[6] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[7] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[8] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[9] has no pins on routing grid
```
Is anything else should be followed in manual macro placement to make DRC clean routing?

osamahammad21, 2023-08-27:
It doesn't look like the pin is on M1 nor M5

vijayank88, 2023-08-28:
@osamahammad21 
After many iterations and above notes in mind, found right co-ordinates to complete the manual macro placement and detail routing completes successfully without DRC violations.
Now I get AP with `OnGrid/OnCenter`
[image]

Find the attached updated detail route test case. Still I am seeing `WARING DRT-0418`
```
[WARNING DRT-0418] Term dmem.dmem0/rd_out[0] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[1] has no pins on routing grid
[WARNING DRT-0418] Term dmem.dmem0/rd_out[2] has no pins on routing grid
```
DRT with no DRC violations, updated test case: https://slack-files.com/T01699QAZBQ-F05PS2PMXH9-f7a160e8e6

osamahammad21, 2023-10-17:
I am closing this issue since DRT finishes without DRC. if the warnings still persist, please open a separate issue for that.

closed 2023-10-17 by osamahammad21