The-OpenROAD-Project/OpenROAD-flow-scripts #919: Unconnected PDN node on net VDD

yupferris, 2023-03-23:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

In https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/918, I'm attempting to clean up floorplanning setup for the `mock-array-big` example. In doing so, I either get DRC errors as power straps end up too close to macros, or, if I adjust the pdn config [as such](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/918/files#diff-27ac0f3dd04316cbe20c08153c07233c020eae57be8e3711d0f220e13af23463L16-R17) to align better with the macros, I get a bunch of `Unconnected PDN node on net VDD` errors, which I don't know how to fix.

### Expected Behavior

The flow completes successfully and generates GDSII.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.0-60-generic
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.22.1
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-7229-gd76c80d8e
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
[... 98 lines cut ...]
- `cd flow`
- `make DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk`

### Relevant log output

```shell
< many similar lines to those below >

[WARNING PSM-0038] Unconnected PDN node on net VDD at location (456.978um, 469.437um), layer: 6.
[WARNING PSM-0039] Unconnected instance ces_6_6 at location (456.978um, 469.437um) layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net VDD at location (456.978um, 463.437um), layer: 5.
[WARNING PSM-0039] Unconnected instance ces_6_6 at location (456.978um, 463.437um) layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net VDD at location (456.978um, 463.437um), layer: 6.
[WARNING PSM-0039] Unconnected instance ces_6_6 at location (456.978um, 463.437um) layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net VDD at location (456.978um, 457.437um), layer: 5.
[WARNING PSM-0039] Unconnected instance ces_6_6 at location (456.978um, 457.437um) layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net VDD at location (456.978um, 457.437um), layer: 6.
[WARNING PSM-0039] Unconnected instance ces_6_6 at location (456.978um, 457.437um) layer: 6.
[WARNING PSM-0038] message limit reached, this message will no longer print
[WARNING PSM-0039] message limit reached, this message will no longer print
[ERROR PSM-0078] IR drop setup failed.  Analysis can't proceed.
Error: final_report.tcl, 58 PSM-0078
Elapsed time: 0:59.33[h:]min:sec. CPU time: user 58.51 sys 0.74 (99%). Peak memory: 1708464KB.
make: *** [logs/asap7/mock-array-big/base/6_report.log] Error 1
```

gadfort, 2023-03-23:
> The DRC errors might relate to https://github.com/The-OpenROAD-Project/OpenROAD/pull/3064 
> 
> @gadfort do you have time to look at the pdn issue?

@maliberty later today possibly, but I don't see a reproducible test case attached.

maliberty, 2023-03-23:
@gadfort just instructions to reproduce above in "To Reproduce".

@yupferris it is faster for developers if you package a standalone test case

gadfort, 2023-03-24:
@maliberty was this resolved with the changes to PDN?

maliberty, 2023-03-24:
I haven't checked yet.

gadfort, 2023-03-24:
@maliberty I tried running it, but I didn't get any errors related to the PDN (assuming I ran the right thing). But the routing finished with 10 dec violations.

yupferris, 2023-03-24:
I've followed the instructions given while creating an issue to create a reproduction archive, with `ISSUE_TAG=unconnected_pdn_node` and `make final_report_issue` as I believe the errors above are generated when running that particular script, even though the issue is likely introduced in pdn: [final_report_unconnected_pdn_node.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11063423/final_report_unconnected_pdn_node.tar.gz)

Alternatively, if it makes any difference, here's an archive created using `make pdn_issue`: [pdn_unconnected_pdn_node.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11063431/pdn_unconnected_pdn_node.tar.gz)

yupferris, 2023-03-24:
Also, the flow does generate `flow/reports/asap7/mock-array-big/base/5_route_drc.rpt` and `flow/reports/asap7/mock-array-big/base/congestion.rpt`, but these files are empty.

gadfort, 2023-03-24:
Okay, so I ran it with the changes, so maybe the fixes fixed the power grid, but it is clearly making it hard for the router to converge.

yupferris, 2023-03-27:
> Okay, so I ran it with the changes, so maybe the fixes fixed the power grid, but it is clearly making it hard for the router to converge. (I ran the test yesterday, and did not have time to look at the uploaded testcase)

I can also see that the design now spends much more time in DRT, but I have no idea why. Are there any tips/further information that can be provided as to why it's working so much harder? Maybe something I can do differently here?

maliberty, 2023-03-27:
@gadfort I still see unconnected PDN nodes with the head of master.  I'll look at the router but pdn still needs help.

gadfort, 2023-03-27:
I'll try to take a closer look tonight

maliberty, 2023-03-27:
@gadfort Thanks.  I wonder if you have thoughts on how the power grid should be constructed.  I believe the block and the top use the same pdn config.  I'm wondering if your prior experience suggests that we should do something differently.

gadfort, 2023-03-27:
@maliberty I usually use a different one for blocks, since you might need to push the grid down to enable power grids later.
Here, it looks like you could juts use M5 as the top level in the block and then M6 will connect over and ensure connectivity.

https://github.com/siliconcompiler/zerosoc/blob/main/openroad/pdngen_core_only.tcl

https://github.com/siliconcompiler/zerosoc/blob/main/openroad/pdngen_top.tcl

maliberty, 2023-03-29:
@gadfort have you had a chance to look at this?

maliberty, 2023-03-29:
> https://github.com/siliconcompiler/zerosoc/blob/main/openroad/pdngen_core_only.tcl
> https://github.com/siliconcompiler/zerosoc/blob/main/openroad/pdngen_top.tcl

@gadfort these examples seem to be hard coded for sky130.  Is this correct?  Do you have asap7 equivalents?

maliberty, 2023-03-29:
@gadfort In #933 I have a solution by having a separate pdn for the Element.  I'm still curious why pdn fails in this configuration (I'm guessing is the blockages).  The priority is lower though.

gadfort, 2023-03-31:
> @gadfort In #933 I have a solution by having a separate pdn for the Element. I'm still curious why pdn fails in this configuration (I'm guessing is the blockages). The priority is lower though.

@maliberty yes, the blockages prevent PDN from doing anything. If there were no blockages on m6 and pins were placed on m5, then PDN could have reached in and connected. Long term I would like to be able to do edge connections to macros (but that also wouldn't apply here.

gadfort, 2023-03-31:
I don't think so, it's on my wishlist to make that work, but at the moment there's no code that would support that

closed 2023-03-31 by maliberty