The-OpenROAD-Project/OpenROAD-flow-scripts #924: Facing error in floorplan of RePlace diverged at initial iteration. Re-run with a smaller init_density_penalty value

VanshikaTanwar, 2023-03-25:
### Subject

[Stage]: Floorplaning.

### Describe the bug

When I am running my design using sky130hd then it's showing error in floorplan step and as I have tried adding and changing the different values of init density penalty and also, place_denaitu but still same error is coming 
Of 
[ERROR GPL-0304] Replace diverged at initial iteration . Re-run with a smaller init_density_penalty value .

### Expected Behavior

Have tried all solutions but not able to solve this error 
Please help me to solve this error .

### Environment

```shell
Linux (Ubuntu 20.04 ) version 
Python 3.8 version
```

### To Reproduce

...

### Screenshots

[image]

Config.mk pic 

[image]

[image]

tdms_place.tcl
[image]

### Additional Context

Please help me how to resolve this error ?

vijayank88, 2023-03-25:
Run `make floorplan_issue` and attach the test case.
Screenshot won't help to fix the issue.

VanshikaTanwar, 2023-03-25:
floorplan is not able to execute because there is no command present in top level makefile of my design 

[image]

maliberty, 2023-03-25:
The screen shots suggest he is using openfasoc.  @msaligane is there a way to package an issue from fasoc?

gadfort, 2023-03-25:
Looking at the density it's extremely sparse at 0.8%, maybe that is contributing to the issue

VanshikaTanwar, 2023-03-26:
So, what should I do now sir ,to solve this error ?

maliberty, 2023-03-26:
You need to find a way to produce a standalone test case.  Without one nothing can be done.  I don't use fasoc so I have no idea how to do so with those scripts.

Eyantra698Sumanto, 2023-03-26:
Hi all, I see a similar issue here: https://github.com/The-OpenROAD-Project/OpenROAD/issues/1296

maliberty, 2023-03-26:
@Eyantra698Sumanto #1296 was closed a long time ago.  How is that related?

Eyantra698Sumanto, 2023-03-26:
@maliberty The above screenshot and the #1296 both show the "RePlace diverged at the initial iteration" hence I was wondering.

antonblanchard, 2023-03-27:
FYI while I think the fix in the PR is wrong, the test case in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2370 still reproduces. In that case the issue is explained in: https://github.com/The-OpenROAD-Project/OpenROAD/pull/2370#issuecomment-1294796770

vijayank88, 2023-03-27:
@antonblanchard 
I have tried with ORFS/OpenROAD commit, it passes. Tail of logs:
```
[InitialPlace]  Iter: 1 CG residual: 0.00000000 HPWL: 912358
[InitialPlace]  Iter: 2 CG residual: 0.00000000 HPWL: 619978
[InitialPlace]  Iter: 3 CG residual: 0.00000001 HPWL: 619978
[InitialPlace]  Iter: 4 CG residual: 0.00000001 HPWL: 619978
[InitialPlace]  Iter: 5 CG residual: 0.00000001 HPWL: 619978
[INFO GPL-0031] FillerInit: NumGCells: 45362
[INFO GPL-0032] FillerInit: NumGNets: 13
[INFO GPL-0033] FillerInit: NumGPins: 25
[INFO GPL-0023] TargetDensity: 0.60
[INFO GPL-0024] AveragePlaceInstArea: 131220
[INFO GPL-0025] IdealBinArea: 218700
[INFO GPL-0026] IdealBinCnt: 45534
[INFO GPL-0027] TotalBinArea: 9958417020
[INFO GPL-0028] BinCnt: 128 128
[INFO GPL-0029] BinSize: 781 779
[INFO GPL-0030] NumBins: 16384
[NesterovSolve] Iter: 1 overflow: 0 HPWL: 605184
[NesterovSolve] Finished with Overflow: 0.000000
```

msaligane, 2023-03-29:
> The screen shots suggest he is using openfasoc. @msaligane is there a way to package an issue from fasoc?

@maliberty we do not have that yet. Maybe something to add @lulu9312

lulu9312, 2023-03-29:
@msaligane Correct we don't have this generator yet. @VanshikaTanwar In second screenshot, for your config.mk file I see PLACE_DENSITY is commented. Maybe you should try uncommenting it. Also to run the `make floorplan_issue` command you need to go flow/ directory.

VanshikaTanwar, 2023-03-29:
@lulu9312 Sir, I already tried uncommenting that PLACE_DENSITY and assigned different values to it but still the same error persist . 

Ok sir I will run the make floorplan_issue and then share the SS /file of it what it generates after this command .

vijayank88, 2023-07-17:
@VanshikaTanwar Is that resolved?

closed 2026-04-17 by github-actions[bot]