The-OpenROAD-Project/OpenROAD-flow-scripts #938: Manual macro placement configuration not read by RTLMP2

vijayank88, 2023-03-30:
### Subject

[Stage]: Floorplaning.

### Describe the bug

Based fixed macro location with `macro.cfg` RTLMP flow not placing in right co-ordinates.

### Expected Behavior

Macros should be placed based on `export MACRO_PLACEMENT = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/macro.cfg` 

### Environment

```shell
with latest master
```

### To Reproduce

Due to file size limit uploaded here: https://drive.google.com/file/d/1Rvno9adoQkzrV6ewoT_LXogtc1EgWzr4/view?usp=share_link

maliberty, 2023-03-31:
If you manually place the macros then you shouldn't be using rtlmp.

vijayank88, 2023-04-01:
Thanks @maliberty.

We have requested documentation for RTLMP tool from @ravi-varadarajan.

The issues was reported by Jignesh from USC_ext.

ravi-varadarajan, 2023-04-01:
This is an enhancement request. mpl2 should be able to handle cases where some of the macros are pre-placed

maliberty, 2023-04-01:
The issue is that orfs doesn't call mpl2 when you give a config file as it assumes all macros are placed.

ravi-varadarajan, 2023-04-01:
This check to read the manual macro placement is only for TMP and not for rtlmp2. I will add it to mpl2  too in  macro_place.tcl.  However we will need to support some of the macros being pre-placed in the flow. I will create an enhancement request and assign it to me.

vijayank88, 2023-08-17:
@ravi-varadarajan 
Any update on this?

maliberty, 2023-08-17:
There are two different things here
- MACRO_PLACEMENT_TCL/MACRO_PLACEMENT should be used even if RTLMP_FLOW is set which is a trivial change.  I can do this.
- mpl2 should support partial macro placement which is probably a lot of work.  This will be an enhancement request to OR.  @ravi-varadarajan did you open one as you said?