The-OpenROAD-Project/OpenROAD-flow-scripts #940: PDN insertion is slow

oharboe, 2023-03-31:
### Description

Run:

```
MOCK_ARRAY_WIDTH=16 MOCK_ARRAY_HEIGHT=8 MOCK_ARRAY_DATAWIDTH=64 MOCK_ARRAY_PICTH_SCALE=20 make DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk floorplan
```

### Suggested Solution

Optimize PDN so it runs faster.

oharboe, 2023-03-31:
@maliberty @tspyrou I've created and verified that the test-case linke to in this github issue can reproduce slow PDN insertion and .lib file creation. In both cases it looks like O(n^2) algorithms.

gadfort, 2023-03-31:
@oharboe the PDN issue is the same as you identified before. It shouldn't be too difficult to go though and correct so the obstruction tree is only built once, but would just take a little bit of time on my part.

openroadie, 2023-04-24:
@oharboe I believe this is handled by the latest STA integration. Could you please confirm ?

oharboe, 2023-04-24:
> @oharboe the PDN issue is the same as you identified before. It shouldn't be too difficult to go though and correct so the obstruction tree is only built once, but would just take a little bit of time on my part.

Yes. Thank you!

closed 2023-04-24 by oharboe