The-OpenROAD-Project/OpenROAD-flow-scripts #944: ERROR: Duplicate MACRO name: Element (inside MACRO) (line=7, cell=, file=Element.lef) in Layout.read

oharboe, 2023-04-03:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

Creating a mock abstracts from mock-array-big fails

### Expected Behavior

Should create an abstract for mock-array-big

### Environment

```shell
Git commit: 47991be874e4db0197c787e072ad23100b5b0911
kernel: Linux 5.19.0-38-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
```

### To Reproduce

```
make DESIGN_CONFIG=designs/asap7/mock-array-big/Element/config.mk skip_cts skip_route generate_abstract
make DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk skip_cts skip_route generate_abstract
[deleted]
ERROR: Duplicate MACRO name: Element (inside MACRO) (line=7, cell=, file=Element.lef) in Layout.read
[deleted]
```

maliberty, 2023-04-03:
I looked at this briefly and it seems like the same lef was loaded twice.

oharboe, 2023-04-03:
> I looked at this briefly and it seems like the same lef was loaded twice.

I *believe* this doesnt happen if you first delete objects and results from my other work today, but I didnt verify on mock-array-big

gkamendje, 2023-06-10:
I am getting this error message as well after cleaning everything with `make clean_all` and subsequently running `make finish`