The-OpenROAD-Project/OpenROAD-flow-scripts #946: Make AutoTuner variable list as tuneable with ORFS flow script

vijayank88, 2023-04-03:
### Description

AutoTuner generating best parameter for the design based on `config.mk`.

`autotuner-best-faf46f4a.json` for gcd design with sky130hd.
```
{
    "_SDC_CLK_PERIOD": 2.303869923183397,
    "CORE_UTILIZATION": 43,
    "CORE_ASPECT_RATIO": 1.8722358051533017,
    "CORE_MARGIN": 2,
    "CELL_PAD_IN_SITES_GLOBAL_PLACEMENT": 2,
    "CELL_PAD_IN_SITES_DETAIL_PLACEMENT": 1,
    "_FR_LAYER_ADJUST": 0.18212956289715632,
    "PLACE_DENSITY_LB_ADDON": 0.07539273004399097,
    "_PINS_DISTANCE": 2,
    "CTS_CLUSTER_SIZE": 33,
    "CTS_CLUSTER_DIAMETER": 383,
    "_FR_GR_OVERFLOW": 1,
    "best_result": 304.42900000000003
}
```
The configuration begin with `_` like `_PINS_DISTANCE`, `_SDC_CLK_PERIOD` etc., can't directly added into `config.mk` to achieve the best PPA as per AutoTuner.

### Suggested Solution

Integrate those variable part of the flow, hence it improves overall quality of flow as well.

Or else point the respective script file (Kind of reference document) to pass above values to improve the QoR. User can go and edit respective script to update from its default value to match AutoTuner results with ORFS flow.

vvbandeira, 2023-04-03:
Variables that start with an underscore are settings that need to be changed in files (e.g., `fastroute.tcl` and `constraints.sdc`). 

For now, documenting is the best path.
But I think the best long-term approach would be to replace `config.mk` files in ORFS with `config.json` files and adapt the flow to read JSON. Also, we are considering moving ORFS to depend on SiliconCompiler, which could fix this issue.

vijayank88, 2023-04-04:
Ok. Thanks @vvbandeira . 
I will add documentation for updating those variable. Once automated, we will disable the document.

luarss, 2025-02-16:
Resolved with addition of `variables.yaml` and https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2640