The-OpenROAD-Project/OpenROAD-flow-scripts #950: Dependencyinstaller doesn't follow semantics of "-common" and "-base" in tools/OpenROAD/etc/DependencyInstaller.sh.

oharboe, 2023-04-03:
### Subject

[Documentation] for documentation errors.

### Describe the bug

Dependencyinstaller doesn't follow semantics of "-common" in tools/OpenROAD/etc/DependencyInstaller.sh.

Only "-base" option should install all the packages requiring sudo.

The following should not require sudo access, but only install components into the folder specified:

```
etc/DependencyInstaller.sh -common -prefix=./dependencies
[deleted]
+ apt-get -y update
Reading package lists... Done
E: Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)
E: Unable to lock directory /var/lib/apt/lists/
W: Problem unlinking the file /var/cache/apt/pkgcache.bin - RemoveCaches (13: Permission denied)
W: Problem unlinking the file /var/cache/apt/srcpkgcache.bin - RemoveCaches (13: Permission denied)
```

### Expected Behavior

Follow -base and -common semantics of tools/OpenROAD/etc/DependencyInstaller.sh

### Environment

```shell
Git commit: 47991be874e4db0197c787e072ad23100b5b0911
kernel: Linux 5.19.0-38-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
```

### To Reproduce

Use procedure above

vijayank88, 2023-04-03:
@oharboe 
Can you try these [options](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/18087e3b5fbd26e712871437c22642720a7aafb7/etc/DependencyInstaller.sh#L99) ?

oharboe, 2023-04-03:
> @oharboe Can you check these [options](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/18087e3b5fbd26e712871437c22642720a7aafb7/etc/DependencyInstaller.sh#L99) ?
> 
> As per existing setup it is applicable only to `OpenROAD` package installer.

I did try these options, they are not implemented according to the semantics in the documentation.

What is needed is a way to distinguish between what needs to be installed as the user(installed into local user folder) and what needs to be installed as root(packages)

tools/OpenROAD/etc/DependencyInstaller.sh "-base" and "-common", provides this capability.

This capability is missing in ORFS/etc/DependencyInstaller.sh

oharboe, 2023-04-03:
> Currently, the only dependency that can be installed as the user (i.e., with `-common`) are `pip` packages as seen [here](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/etc/DependencyInstaller.sh#L19-L25). I don't think that we need any additional logic at the moment, I am missing something?

See error message if you try to install user.components in the description of this issue

closed 2023-04-11 by vvbandeira