The-OpenROAD-Project/OpenROAD-flow-scripts #952: Docker on ubuntu--groups: cannot find name for group

hanm2019, 2023-04-04:
### Description

build a docker on ubuntu and then run it with `docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow openroad/flow-$OS_NAME-builder`

get: 
```
groups: cannot find name for group ID 1003
I have no name!@9a41a4f15c50:/OpenROAD-flow-scripts$ echo $USER

I have no name!@9a41a4f15c50:/OpenROAD-flow-scripts$ whoami
whoami: cannot find name for user ID 1003
I have no name!@9a41a4f15c50:/OpenROAD-flow-scripts$ id
uid=1003 gid=1003 groups=1003
```

### Suggested Solution

update the docker start cmd

 from 
```
docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow openroad/flow-$OS_NAME-builder
```

to 

```
docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow  -v /etc/passwd:/etc/passwd:ro -v /etc/group:/etc/group:ro openroad/flow-$OS_NAME-builder:latest

```

this is just for ubuntu :)

vijayank88, 2023-04-04:
@hanm2019 
How you installed OpenROAD-flow-scripts in docker mode?
Did you followed this instruction https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildWithDocker.html ?

vijayank88, 2023-04-04:
Before running docker command set following in terminal if default steps used for installation:
`export OS_NAME=centos7`

hanm2019, 2023-04-04:
> 

Yes, I install it in ubuntu

hanm2019, 2023-04-04:
> 

I run in ubuntu OS, so I set the OS_NAME to `ubuntu20.04` by

```
export OS_NAME=ubuntu20.04
```

vijayank88, 2023-04-04:
After `export OS_NAME=ubuntu20.04` did you followed below steps for installation?
```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./etc/DockerHelper.sh create -target=dev -os=$OS_NAME
./etc/DockerHelper.sh create -target=builder -os=$OS_NAME
```

hanm2019, 2023-04-04:
> After `export OS_NAME=ubuntu20.04` did you followed below steps for installation?
> 
> ```
> git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
> cd OpenROAD-flow-scripts
> ./etc/DockerHelper.sh create -target=dev -os=$OS_NAME
> ./etc/DockerHelper.sh create -target=builder -os=$OS_NAME
> ```
> 
> Otherwise docker default built with `centos7` OS

Yes, and my host OS is ubuntu18.04, but I config the `OS_NAME` to `ubuntu20.04` and after running
```
./etc/DockerHelper.sh create -target=dev -os=$OS_NAME
./etc/DockerHelper.sh create -target=builder -os=$OS_NAME

```
I build the image successfully.

rovinski, 2023-04-04:
@hanm2019 is this on a machine which has a networked login? e.g. the passwd and group files are stored on the network?

hanm2019, 2023-04-04:
I build the docker image in a machine named A and then transfer the image to another machine named B. the `groups: cannot find name for group ID 1003` is shown when I run the docker on machine B.

hanm2019, 2023-04-04:
the image is `by1906007/flow-ubuntu20.04-builder:latest`

rovinski, 2023-04-04:
Are the `passwd` and `groups` files different between machines `A` and `B`? It doesn't really feel like mounting the passwd and groups files is the right solution here - a vast majority of users will never encounter this issue...

hanm2019, 2023-04-05:
Yeah, the `passwd` and `group` files are different between machine `A` and `B`.

vijayank88, 2023-04-05:
> I build the docker image in a machine named A and then transfer the image to another machine named B. the `groups: cannot find name for group ID 1003` is shown when I run the docker on machine B.

@hanm2019 above should be part of your issue description. So it will easy to understand the problem and resolved quickly.
Without proper description it leads to long conversation.

vijayank88, 2023-04-08:
@vvbandeira 
Is mirror copying will work ?