The-OpenROAD-Project/OpenROAD-flow-scripts #965: Issue with power stripe of uart_rx

vijayank88, 2023-04-07:
### Subject

[Stage]: Floorplaning.

### Describe the bug

I am trying to [uart_rx](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/designs/asap7/uart-blocks/uart_rx/config.mk) but only `M5` vertical stripe created. `M4` horizontal layer not created for this design.

It causing failure in PSM, if I enable macro grid connection
```
[WARNING PSM-0038] Unconnected PDN node on net VDD at location (14.563um, 32.142um), layer: 4.

[WARNING PSM-0039] Unconnected instance uart_rx_inst at location (14.563um, 32.142um) layer: 4.

[WARNING PSM-0038] Unconnected PDN node on net VDD at location (14.563um, 32.142um), layer: 5.

[WARNING PSM-0039] Unconnected instance uart_rx_inst at location (14.563um, 32.142um) layer: 5.

[ERROR PSM-0078] IR drop setup failed.  Analysis can't proceed.

Error: final_report.tcl, 58 PSM-0078
```

### Expected Behavior

Resolve unconnected pdn issue.

### Environment

```shell
Latest ORFS head
```

### To Reproduce

Due to file size limit test case added here: https://slack-files.com/T01699QAZBQ-F0529SMFARG-1cf30bcfa8
`floorplan_uart-blocks_uart_rx_asap7_rx_2023-04-07_13-07.tar.gz`

maliberty, 2023-04-09:
Its not clear how to reproduce the result shown.  The test case is just a packaged output and not a reproducer.

Is this related to https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/960 ?

vijayank88, 2023-04-10:
Yes. As per your [comment](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/960#issuecomment-1501050053) I tried to connect with M5-M6 also to resolve macro connection.
Still it's fail during IR Drop calculation with `UnConnected PDN node VDD`.

Also I doubt in `uart_rx` not having horizontal power stripe and causing that PSM to fail.

vijayank88, 2023-04-13:
Resolved by #960. Added separate pdn_grid as it uses `BLOCKS`.

closed 2023-04-13 by vijayank88