The-OpenROAD-Project/OpenROAD-flow-scripts #966: the clock unit in constraint.sdc

hanm2019, 2023-04-08:
### Description

It seem that the clock unit between `asap7` and `nangate45` are different

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/a2f687212d672e86108b06f7e15671a885803e80/flow/designs/nangate45/tinyRocket/constraint.sdc#L9

it seem that the 2.0322 represent 2.0322ns

however, 

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/a2f687212d672e86108b06f7e15671a885803e80/flow/designs/asap7/riscv32i/constraint.sdc#L5

it seems that the 1660 represent 1660ps

vijayank88, 2023-04-08:
@hanm2019 
So what is your question here?

hanm2019, 2023-04-08:
maybe we should use the same unit of clock?

vijayank88, 2023-04-08:
Time unit not decided by us. It was fixed by library provider.
We have to follow the library units.
Check the respective technology library file for `time_unit`.

closed 2023-04-08 by vijayank88