The-OpenROAD-Project/OpenROAD-flow-scripts #980: Disk quota exceeded

vijayank88, 2023-04-12:
### Subject

[Build] for compilation-related issues.

### Describe the bug

I am trying to install latest ORFS with Censtos 07.

I have cloned the repo and run `sudo ./setup.sh`. If fails with below error.
```
Installing collected packages: pytz, numpy, six, python-dateutil, pandas
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/pip/basecommand.py", line 215, in main
	status = self.run(options, args)
  File "/usr/lib/python3.6/site-packages/pip/commands/install.py", line 365, in run
	strip_file_prefix=options.strip_file_prefix,
  File "/usr/lib/python3.6/site-packages/pip/req/req_set.py", line 789, in install
	**kwargs
  File "/usr/lib/python3.6/site-packages/pip/req/req_install.py", line 854, in install
	strip_file_prefix=strip_file_prefix
  File "/usr/lib/python3.6/site-packages/pip/req/req_install.py", line 1069, in move_wheel_files
	strip_file_prefix=strip_file_prefix,
  File "/usr/lib/python3.6/site-packages/pip/wheel.py", line 345, in move_wheel_files
	clobber(source, lib_dir, True)
  File "/usr/lib/python3.6/site-packages/pip/wheel.py", line 316, in clobber
	ensure_dir(destdir)
  File "/usr/lib/python3.6/site-packages/pip/utils/__init__.py", line 83, in ensure_dir
	os.makedirs(path)
  File "/usr/lib64/python3.6/os.py", line 220, in makedirs
	mkdir(name, mode)
OSError: [Errno 122] Disk quota exceeded: '/home/vijayan/.local/lib/python3.6/site-packages/pandas/_libs/tslibs'
```

@vvbandeira @oharboe any suggestion why it fails?

`/home/vijayan` has `4.5T` free space.

### Expected Behavior

Pass `setup.sh` script

### Environment

```shell
[vijayan@vlsi09 OpenROAD]$ ./etc/Env.sh 
mkdir: cannot create directory ‘build’: Disk quota exceeded
```

### To Reproduce

`sudo ./setup.sh` in Centos07

oharboe, 2023-04-12:
@vijayank88 What do you think this error means above? ```Disk quota exceeded```

vijayank88, 2023-04-12:
Yes, it about local disk allocated. I have informed to IT team about the error and waiting for the response.

Meanwhile am looking for arguments to install packages with custom location.

Is that possible with `setup.sh`?

vvbandeira, 2023-04-12:
@vijayank88 is this a GCP machine or a machine from Chipware? If it is from Chipware is possible that they have a limit per user which you might be exceeding. Run `du -hd0 /home/vijayan`, if it is a round number is likely this is the problem.

vijayank88, 2023-04-12:
@vvbandeira 
I have messaged you in Slack channel DM. Please check that.

oharboe, 2023-04-12:
@vijayank88 Glad to see you are using the new setup procedure thought :-) It is intended to be a no decisions, no questions, no harm done to your machine setup procedure.

vijayank88, 2023-04-12:
As per update from @vvbandeira , it is purely related to allocated disk space for the user. Hence closing this. Thanks @oharboe

closed 2023-04-12 by vijayank88